Joshua-Riek/ubuntu-rockchip #1: usb tethering not working

develperbayman, 2023-02-07:
simply the usb tethering isnt working

develperbayman, 2023-02-07:
> Hello, I'm unfamiliar with USB tethering, is that when you use your phone
> as a hotspot and connect it to a device with a cable for internet?

Yes usually default behavior is automatic on device plug in very important
because orange pi 5 ships without wifi

Joshua-Riek, 2023-02-07:
I see, I will take a look at this when I have time later today!

Joshua-Riek, 2023-02-08:
I tested USB tethering out on the Ubuntu 20.04 and 22.04 desktop images in the v1.2 release, and it worked without an issue. I did make some recent changes fixing some wifi and ethernet issues that will be included in the next release, v1.3. I plan on uploading that release sometime tonight or tomorrow. Look out for the v1.3 release and perhaps with the ethernet changes I made, USB tethering will work properly for you.

develperbayman, 2023-02-09:
yes totally works!

closed 2023-02-09 by develperbayman