Joshua-Riek/ubuntu-rockchip #2: Broadcom AP6275P find no wifi-networks

lowelowelowe, 2023-02-07:
Broadcom AP6275P find no wifi-networks

I use m.2 wifi module from official orange pi5 ,the "ap6275p"

lshw showed the driver properly installed but cannot find any wifi network to join,
in the settings -> wifi "Visible networks" keep spinning. not a single wifi network show.

Joshua-Riek, 2023-02-07:
Hey! I recently tested that m.2 wifi module myself and found the same issue. I need to figure out why the settings won't find any networks. I have found that connecting to a wifi network through the command line will work correctly. 

To connect to a wifi network through the command line:

1. Edit the file /etc/wpa_supplicant/wpa_supplicant.conf and enter your wifi credentials. The file should look like the one below.
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="your_wifi_name"
    psk="your_wifi_password"
    key_mgmt=WPA-PSK
    priority=1
}
```
2. Reboot the system.

Joshua-Riek, 2023-02-07:
I found a solution and now I can see the list of available wifi networks in the wifi settings menu. Looks like I need to use NetworkManager instead of wpa_supplicant. I'll be working on the changes I need to make this afternoon.

lowelowelowe, 2023-02-07:
> Hey! I recently tested that m.2 wifi module myself and found the same issue. I need to figure out why the settings won't find any networks. I have found that connecting to a wifi network through the [...]

Thank you very much. 
Wifi work perfectly after edit in  "/etc/wpa_supplicant/wpa_supplicant.conf"
both 2.4ghz and 5ghz.

FYI. Bluetooth on  AP6275P  also not working . Pretty much the same problem like wifi in the gui setting.
When I tried "bluetoothctl show" in terminal
-> No default controller available

cheers.

Joshua-Riek, 2023-02-08:
I just fixed this issue with the wifi in the commit https://github.com/Joshua-Riek/ubuntu-orange-pi5/commit/a36c1891bce513ac1f80b4ab2043c8f62d1fa8c6.
I'll be creating a separate issue to get Bluetooth working properly.

closed 2023-02-08 by Joshua-Riek