Joshua-Riek/ubuntu-rockchip #4: hdmi not connected before login screen disables wayland

Joshua-Riek, 2023-02-08:
If you power on your Orange Pi 5 with either desktop image without plugging in an HDMI cable the system will only use X11.

develperbayman, 2023-02-08:
one thing i think is weird on every distro i try for the orange pi 5 is i always have to unplug and plug back in my hdmi or it dont register but only after long periods of power off?

Joshua-Riek, 2023-02-08:
Do you mean when the device is fully powered off? Or when the screen automatically turns off and the system is 'asleep'?

develperbayman, 2023-02-09:
after long periods of no power but this might also be a tv thing im hooking it up too is rather old however i cannot get waydroid to run i am on wayland for sure so i dont know if wayland is the culprit this is by far the best orange pi 5 distro iv seen so far i can get retroarch to work which i cannot on any of the other distros  i notice it does not really seem to not like multitasking very much ...for example i like to stream movies on chromium in the background while working on click packages or web/python development stuff everything  makes chromeium hang

Joshua-Riek, 2023-02-09:
With the Ubuntu 22.04 release dragging around windows like chromium can cause lag. And there are weird GPU-related errors the kernel generates. But the Ubuntu 20.04 release is relatively stable from my testing. If you want to create a new issue for the chromium hang and provide more detail so I can recreate I would be happy to take a deeper look, which release are you on? Also, did you look at the new release to see if the USB tethering works for you?

develperbayman, 2023-02-09:
do you think i can compile this on my opi5?? ....if so ill try to fork/pull it ....iv never compiled ubuntu before

Joshua-Riek, 2023-02-09:
Yeah, you should be able to compile on the opi5! Just download all the required packages in the requirements section on the readme. Make sure that you have some cooling on the opi5 since the build does a lot of code compiling. On average, a build will take 40 minutes on my desktop, but I have no clue how long on the opi5. I will have to try firefox out, it should not be working with hardware-accelerated video playback.

develperbayman, 2023-02-09:
[image]

develperbayman, 2023-02-09:
for sure no HWA for firefox cs-go online super slow and no sound ....never did understand that you would think it would be system wide being a kernel driver and all??

Joshua-Riek, 2023-02-09:
To get hardware acceleration for firefox, someone would need to add support to firefox itself. For example, chromium is using a custom version that is built and tweaked in a certain way to enable hardware acceleration. If someone were to uninstall chromium and then re-install with apt-get or snap chromium would no longer have hardware acceleration.

develperbayman, 2023-02-09:
so question if i say write a java app that uses a webview from any library say even chromium would it benefit from hardware acceleration from openGL or whatever interface that uses the system driver acceleration? i guess im not to sure where hardware acceleration is applicable?

Joshua-Riek, 2023-02-09:
I'm not familiar with java and webviews to know for sure. But if the app uses the included chromium browser or uses OpenGL directly for things such as 3d graphics should work with hardware acceleration.

Joshua-Riek, 2023-02-10:
I will be closing this issue and then re-opening it under a new issue; I created a [discussion](https://github.com/Joshua-Riek/ubuntu-orange-pi5/discussions/9) board to keep issues with content relevant to its solution.