Joshua-Riek/ubuntu-rockchip #6: build errors building  arm64,orange pi 5

develperbayman, 2023-02-09:
this is my terminal output:

` DTC     arch/arm64/boot/dts/rockchip/overlay/rk3588-dmc.dtbo
  MKIMAGE arch/arm64/boot/dts/rockchip/overlay/rk3588-fixup.scr
Image Name:   
Created:      Wed Feb  8 23:48:47 2023
Image Type:   AArch64 Linux Script (uncompressed)
Data Size:    2692 Bytes = 2.63 KiB = 0.00 MiB
Load Address: 00000000
Entry Point:  00000000
Contents:
   Image 0: 2684 Bytes = 2.62 KiB = 0.00 MiB
  HOSTLD  scripts/genksyms/genksyms
scripts/extract-cert.c: In function ‘display_openssl_errors’:
scripts/extract-cert.c:46:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   46 |         while ((e = ERR_get_error_line(&file, &line))) {
      |         ^~~~~
In file included from scripts/extract-cert.c:23:
/usr/include/openssl/err.h:411:15: note: declared here
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
      |               ^~~~~~~~~~~~~~~~~~
scripts/extract-cert.c: In function ‘drain_openssl_errors’:
scripts/extract-cert.c:60:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   60 |         while (ERR_get_error_line(&file, &line)) {}
      |         ^~~~~
In file included from scripts/extract-cert.c:23:
/usr/include/openssl/err.h:411:15: note: declared here
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
      |               ^~~~~~~~~~~~~~~~~~
scripts/extract-cert.c: In function ‘main’:
scripts/extract-cert.c:130:17: warning: ‘ENGINE_load_builtin_engines’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  130 |                 ENGINE_load_builtin_engines();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from scripts/extract-cert.c:24:
/usr/include/openssl/engine.h:358:28: note: declared here
  358 | OSSL_DEPRECATEDIN_3_0 void ENGINE_load_builtin_engines(void);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
scripts/extract-cert.c:132:17: warning: ‘ENGINE_by_id’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  132 |                 e = ENGINE_by_id("pkcs11");
      |                 ^
In file included from scripts/extract-cert.c:24:
/usr/include/openssl/engine.h:336:31: note: declared here
  336 | OSSL_DEPRECATEDIN_3_0 ENGINE *ENGINE_by_id(const char *id);
      |                               ^~~~~~~~~~~~
scripts/extract-cert.c:134:17: warning: ‘ENGINE_init’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  134 |                 if (ENGINE_init(e))
      |                 ^~
In file included from scripts/extract-cert.c:24:
/usr/include/openssl/engine.h:620:27: note: declared here
  620 | OSSL_DEPRECATEDIN_3_0 int ENGINE_init(ENGINE *e);
      |                           ^~~~~~~~~~~
scripts/extract-cert.c:139:25: warning: ‘ENGINE_ctrl_cmd_string’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  139 |                         ERR(!ENGINE_ctrl_cmd_string(e, "PIN", key_pass, 0), "Set PKCS#11 PIN");
      |                         ^~~
In file included from scripts/extract-cert.c:24:
/usr/include/openssl/engine.h:479:5: note: declared here
  479 | int ENGINE_ctrl_cmd_string(ENGINE *e, const char *cmd_name, const char *arg,
      |     ^~~~~~~~~~~~~~~~~~~~~~
scripts/extract-cert.c:140:17: warning: ‘ENGINE_ctrl_cmd’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  140 |                 ENGINE_ctrl_cmd(e, "LOAD_CERT_CTRL", 0, &parms, NULL, 1);
      |                 ^~~~~~~~~~~~~~~
In file included from scripts/extract-cert.c:24:
/usr/include/openssl/engine.h:450:27: note: declared here
  450 | OSSL_DEPRECATEDIN_3_0 int ENGINE_ctrl_cmd(ENGINE *e, const char *cmd_name,
      |                           ^~~~~~~~~~~~~~~
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
/usr/bin/env: ‘python’: No such file or directory
make[1]: *** [scripts/Makefile.build:273: scripts/mod/empty.o] Error 127
make[1]: *** Waiting for unfinished jobs....
  CC      scripts/mod/devicetable-offsets.s
/usr/bin/env: ‘python’: No such file or directory
make[1]: *** [scripts/Makefile.build:117: scripts/mod/devicetable-offsets.s] Error 127
make: *** [Makefile:1322: prepare0] Error 2
Error: in ./scripts/build-kernel.sh on line 42
Error: in ./build.sh on line 29
ubuntu@orange-pi:~/Documents/ubuntu-orange-pi5$ 

`

also i have 3 directories in the build folder  linux-orangepi , rkbin ,u-boot-orangepi 

it says the image was created so not do i just put i directory into a img file? .....im kinda learning here also for the build dependencies i had to change grub-pc to grub-common ....once i under stand the whole build >theres my image>try it cycle ill be more help i hope

Joshua-Riek, 2023-02-09:
It looks like python3 is missing; I need to add that to add the package to the readme! Try `sudo apt-get install python3` and start the build again, should work.

Joshua-Riek, 2023-02-09:
Also, I see the issue with the grub-pc package, and it's replaced with `sudo apt-get install grub2-common grub-common`.

develperbayman, 2023-02-09:
python not available? so i did this

ubuntu@orange-pi:~/Documents/ubuntu-orange-pi5$ sudo apt-get install python
[sudo] password for ubuntu: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  2to3 python2-minimal python2 dh-python python-is-python3

E: Package 'python' has no installation candidate
ubuntu@orange-pi:~/Documents/ubuntu-orange-pi5$ sudo apt-get install python2 python2-minimal
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib python2.7 python2.7-minimal
Suggested packages:
  python2-doc python-tk python2.7-doc
The following NEW packages will be installed:
  libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib python2 python2-minimal python2.7 python2.7-minimal
ython

Joshua-Riek, 2023-02-09:
Oh, right, there are different python packages for Ubuntu 20.04 and 22.04. I use 20.04 to build this project and would prefer not to support building on 22.04 at this time just for my own consistency. But installing python2-minimal and python-is-python3 should fix it. I don't know if any other errors will occur, I have not tested a full build with Ubuntu 22.04.

develperbayman, 2023-02-09:
hmm still trowing python errors    but everything installed 

ools/rockchip/boot_merger.c: In function ‘main’:
tools/rockchip/boot_merger.c:895:18: warning: array subscript 20 is outside array bounds of ‘char[20]’ [-Warray-bounds]
  895 |         str[len] = 0;
      |         ~~~~~~~~~^~~
tools/rockchip/boot_merger.c:933:14: note: while referencing ‘name’
  933 |         char name[MAX_NAME_LEN];
      |              ^~~~
  HOSTLD  tools/boot_merger
  CC      arch/arm/cpu/armv8/fwcall.o
  LD      arch/arm/cpu/armv8/built-in.o
  CC      common/main.o
  CC      cmd/version.o
  LD      common/built-in.o
  LD      cmd/built-in.o
  CC      drivers/usb/gadget/f_fastboot.o
  CC      lib/display_options.o
  LD      lib/built-in.o
  LD      drivers/usb/gadget/built-in.o
  LD      u-boot
  OBJCOPY u-boot-nodtb.bin
make[2]: 'arch/arm/dts/rk3588s-orangepi-5.dtb' is up to date.
start=$(aarch64-linux-gnu-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(aarch64-linux-gnu-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x00200000 $start $end
  MKIMAGE u-boot.itb
Image Type:   Rockchip RK35 boot image
Init Data Size: 65536 bytes
Boot Data Size: 225280 bytes
/usr/bin/env: ‘python’: No such file or directory
/usr/bin/env: ‘python’: No such file or directory
/usr/bin/env: ‘python’: No such file or directory
/usr/bin/env: ‘python’: No such file or directory
/usr/bin/env: ‘python’: No such file or directory
init/Kconfig:33:warning: 'GCC_VERSION': number is invalid
#
# configuration written to .config
#
arch/arm64/Makefile:44: Detected assembler with broken .inst; disassembly will be unreliable
/usr/bin/env: ‘python’: No such file or directory
  SYNC    include/config/auto.conf.cmd
/usr/bin/env: ‘python’: No such file or directory
/usr/bin/env: ‘python’: No such file or directory
/usr/bin/env: ‘python’: No such file or directory
/usr/bin/env: ‘python’: No such file or directory
/usr/bin/env: ‘python’: No such file or directory
init/Kconfig:33:warning: 'GCC_VERSION': number is invalid
arch/arm64/Makefile:44: Detected assembler with broken .inst; disassembly will be unreliable
/usr/bin/env: ‘python’: No such file or directory
  CC      scripts/mod/empty.o
/usr/bin/env: ‘python’: No such file or directory
make[1]: *** [scripts/Makefile.build:273: scripts/mod/empty.o] Error 127
make[1]: *** Waiting for unfinished jobs....
  CC      scripts/mod/devicetable-offsets.s
/usr/bin/env: ‘python’: No such file or directory
make[1]: *** [scripts/Makefile.build:117: scripts/mod/devicetable-offsets.s] Error 127
make: *** [Makefile:1322: prepare0] Error 2
make: *** Waiting for unfinished jobs....
Error: in ./scripts/build-kernel.sh on line 42
Error: in ./build.sh on line 29
ubuntu@orange-pi:~/Documents/ubuntu-orange-pi5$ sudo apt-get install python-is-python3
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  python-is-python3
0 upgraded, 1 newly installed, 0 to remove and 20 not upgraded.
Need to get 2,788 B of archives.
After this operation, 13.3 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 python-is-python3 all 3.9.2-2 [2,788 B]
Fetched 2,788 B in 0s (6,745 B/s)             
Selecting previously unselected package python-is-python3.
(Reading database ... 301539 files and directories currently installed.)
Preparing to unpack .../python-is-python3_3.9.2-2_all.deb ...
Unpacking python-is-python3 (3.9.2-2) ...
Setting up python-is-python3 (3.9.2-2) ...
Processing triggers for man-db (2.10.2-1) ...
ubuntu@orange-pi:~/Documents/ubuntu-orange-pi5$

Joshua-Riek, 2023-02-09:
I just tested the build on a fresh install of Ubuntu 22.04 without error, I did the following:
```bash
ubuntu@orange-pi:~$ sudo apt-get install -y build-essential gcc-aarch64-linux-gnu bison \
qemu-user-static qemu-system-arm qemu-efi u-boot-tools binfmt-support \
debootstrap flex libssl-dev bc rsync kmod cpio xz-utils fakeroot parted \
udev dosfstools uuid-runtime git-lfs grub-common python-is-python3
ubuntu@orange-pi:~$ git clone https://github.com/Joshua-Riek/ubuntu-orange-pi5.git
ubuntu@orange-pi:~$ cd ubuntu-orange-pi5
ubuntu@orange-pi:~$ sudo ./build.sh focal
```

Joshua-Riek, 2023-02-10:
Hey, do you still have any issues building?

Joshua-Riek, 2023-02-11:
## How to build on the Ubuntu 22.04 Orange Pi 5 Image

1. Install required packages
```
ubuntu@orange-pi:~$ sudo apt-get install -y build-essential gcc-aarch64-linux-gnu bison \
qemu-user-static qemu-system-arm qemu-efi u-boot-tools binfmt-support \
debootstrap flex libssl-dev bc rsync kmod cpio xz-utils fakeroot parted \
udev dosfstools uuid-runtime git-lfs python-is-python3
```
2. Clone the repo with latest changes
```
ubuntu@orange-pi:~$ git clone https://github.com/Joshua-Riek/ubuntu-orange-pi5.git
ubuntu@orange-pi:~$ cd ubuntu-orange-pi5
```
3. Build
```
ubuntu@orange-pi:~/ubuntu-orange-pi5$ sudo ./build.sh jammy
```

closed 2023-02-11 by Joshua-Riek