Joshua-Riek/ubuntu-rockchip #7: kernel: ashmem and binder? waydroid

develperbayman, 2023-02-09:
https://github.com/waydroid/waydroid/issues/136  similar issue?... different distro

`sudo modprobe binder_linux
modprobe: FATAL: Module ashmem_linux not found in directory /lib/modules/5.10.110-orange-pi
modprobe: FATAL: Module binder_linux not found in directory /lib/modules/5.10.11`

when trying to run waydroid:

`ubuntu@orange-pi:~$ waydroid show-full-ui
[02:43:20] Starting waydroid session
[02:43:20] Failed to start Clipboard manager service, check logs
[gbinder] Service manager /dev/binder has appeared
[gbinder] WARNING: Service manager /dev/binder has died
`

terrified to mod the kernel or mod the kernel with outdated instructions that get a 404 on github

Joshua-Riek, 2023-02-09:
I updated the default kernel config with the commit https://github.com/Joshua-Riek/ubuntu-orange-pi5/commit/29b330bf129cfe9bbcc6296269264407c89e0ac8 to include the following:
```
CONFIG_PSI=y
CONFIG_PSI_DEFAULT_DISABLED=y
```
With these changes to the kernel and adding psi=1 to the kernel command line it works on my end.

Joshua-Riek, 2023-02-09:
Also to add psi=1 to the kernel command line you will need to edit the boot script, I plan on making this a much simpler process in the future, but this will work for now:

```bash
ubuntu@orange-pi:~$ sudo sed -i 's/rootwait rw/rootwait rw psi=1/g' -i /boot/firmware/boot.cmd
ubuntu@orange-pi:~$ sudo mkimage -A arm64 -O linux -T script -C none -n "Boot Script" -d /boot/firmware/boot.cmd /boot/firmware/boot.scr
ubuntu@orange-pi:~$ sudo reboot now
```

develperbayman, 2023-02-09:
so i ran these three commands and this is my output im i cannot really edit anything still trying to get this to compile can i edit the script on the system i have live now?? ....prob have to wait for your next build 

`waydroid show-full-ui
[18:17:51] Waiting for binder Service Manager...
[18:18:50] Service Manager never appeared
[18:18:50] Failed to access IPlatform service
`

Joshua-Riek, 2023-02-09:
Correct either wait until the next build or you could try to build and upgrade the kernel. I can write up a quick example on how to do that in a few minutes.

Joshua-Riek, 2023-02-10:
## How to build and update the kernel on the Ubuntu 22.04 Orange Pi 5 Image

1. Install required packages
```
ubuntu@orange-pi:~$ sudo apt-get install -y build-essential gcc-aarch64-linux-gnu bison \
qemu-user-static qemu-system-arm qemu-efi u-boot-tools binfmt-support \
debootstrap flex libssl-dev bc rsync kmod cpio xz-utils fakeroot parted \
udev dosfstools uuid-runtime git-lfs grub-common python-is-python3
```
2. Clone the repo with latest changes
```
ubuntu@orange-pi:~$ git clone https://github.com/Joshua-Riek/ubuntu-orange-pi5.git
ubuntu@orange-pi:~$ cd ubuntu-orange-pi5
```
3. Build the kernel
```
ubuntu@orange-pi:~/ubuntu-orange-pi5$ sudo ./scripts/build-kernel.sh
```
4. Install the kernel 
```
ubuntu@orange-pi:~/ubuntu-orange-pi5$ sudo dpkg -i build/*.deb
ubuntu@orange-pi:~/ubuntu-orange-pi5$ sudo depmod -a
```
5. Reboot to apply changes
```
ubuntu@orange-pi:~/ubuntu-orange-pi5$ sudo reboot now
```

Joshua-Riek, 2023-02-11:
I enabled support for this by default in the kernel. See commit 7bd93a45b0bf41316a7fcfbeb20fdfd8517d1fc5. This should work properly without any modification in the new release [v1.4](https://github.com/Joshua-Riek/ubuntu-orange-pi5/releases/tag/v1.4). Please open this issue again if there are any other problems.

Joshua-Riek, 2023-02-12:
## How to setup and install waydroid

Note, I'm editing waydroid_base.prop to disable hardware acceleration. That seemed to be the original issue. I will need to figure out what options are compatible with the Orange Pi5 hardware. 

1. Install pre-requisites
```
sudo apt install curl ca-certificates -y
```

2. Add the official repository
```
curl https://repo.waydro.id | sudo bash
```

3. Install waydroid
```
sudo apt install waydroid -y
```

4. Initialize waydroid
```
sudo waydroid init
```

5. Edit `/var/lib/waydroid/waydroid_base.prop` and set:
```
ro.hardware.gralloc=default
ro.hardware.egl=swiftshader
```

6. Restart waydroid container
```
sudo systemctl restart waydroid-container.service
```

7. Start waydroid session 
```
waydroid session start
```

8. In a diffrent terminal start waydroid in full screen mode
```
waydroid show-full-ui
```

Joshua-Riek, 2023-02-12:
When gbm and mesa are in waydroid_base.prop I see the below log from the `waydroid logcat` command.
It looks like mesa can't start properly and cannot find rknpu_dri.so.

```
02-12 04:16:07.618   505   505 D RenderEngine: RenderEngine GLES Backend
02-12 04:16:07.621   505   505 D libEGL  : loaded /vendor/lib64/egl/libEGL_mesa.so
02-12 04:16:07.622   505   505 D libEGL  : loaded /vendor/lib64/egl/libGLESv1_CM_mesa.so
02-12 04:16:07.623   505   505 D libEGL  : loaded /vendor/lib64/egl/libGLESv2_mesa.so
02-12 04:16:07.626   505   505 W EGL-MAIN: MESA-LOADER: failed to open rknpu: dlopen failed: library "/vendor/lib64/dri/rknpu_dri.so" not found (search paths /vendor/lib64/dri, suffix _dri)
02-12 04:16:07.626   505   505 W EGL-MAIN: Failed to open any DRM device
02-12 04:16:07.627   505   505 W EGL-MAIN: Failed to open any DRM device
02-12 04:16:07.627   505   505 W libEGL  : eglInitialize(0xb400007f94840000) failed (EGL_NOT_INITIALIZED)
02-12 04:16:07.627   505   505 E libEGL  : call to OpenGL ES API with no current context (logged once per thread)
02-12 04:16:07.627   505   505 F RenderEngine: eglQueryStringImplementationANDROID(EGL_VERSION) failed
02-12 04:16:07.627    50    50 I tombstoned: received crash request for pid 505
02-12 04:16:07.627    50    50 I tombstoned: enqueueing crash request for pid 505
02-12 04:16:07.656   416   439 I system_server: libdebuggerd_client: done dumping process 161
02-12 04:16:07.657   505   505 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-12 04:16:07.657   505   505 F DEBUG   : LineageOS Version: '18.1-20230211-VANILLA-waydroid_arm64'
02-12 04:16:07.657   505   505 F DEBUG   : Build fingerprint: 'waydroid/lineage_waydroid_arm64/waydroid_arm64:11/RQ3A.211001.001/20:userdebug/test-keys'
02-12 04:16:07.657   505   505 F DEBUG   : Revision: '0'
02-12 04:16:07.657   505   505 F DEBUG   : ABI: 'arm64'
02-12 04:16:07.657   505   505 F DEBUG   : Timestamp: 2023-02-12 04:16:07+0000
02-12 04:16:07.657   505   505 F DEBUG   : pid: 505, tid: 505, name: surfaceflinger  >>> /system/bin/surfaceflinger <<<
02-12 04:16:07.657   505   505 F DEBUG   : uid: 1000
02-12 04:16:07.657   505   505 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
02-12 04:16:07.657   505   505 F DEBUG   : Abort message: 'eglQueryStringImplementationANDROID(EGL_VERSION) failed'
02-12 04:16:07.657   505   505 F DEBUG   :     x0  0000000000000000  x1  00000000000001f9  x2  0000000000000006  x3  0000007fc52b6170
02-12 04:16:07.657   505   505 F DEBUG   :     x4  0000000063e86807  x5  0000000063e86807  x6  0000000063e86807  x7  000000000006db4c
02-12 04:16:07.657   505   505 F DEBUG   :     x8  00000000000000f0  x9  be52aeebddc5c4d3  x10 0000000000000000  x11 ffffffc0ffffffdf
02-12 04:16:07.657   505   505 F DEBUG   :     x12 0000000000000001  x13 0000000000000038  x14 ff00000000000000  x15 0000000000000000
02-12 04:16:07.658   505   505 F DEBUG   :     x16 0000007f95585948  x17 0000007f95564410  x18 0000007f972e0000  x19 00000000000001f9
02-12 04:16:07.658   505   505 F DEBUG   :     x20 00000000000001f9  x21 00000000ffffffff  x22 0000000300000001  x23 0000000000000001
02-12 04:16:07.658   505   505 F DEBUG   :     x24 0000000000000003  x25 0000000000000000  x26 0000000000000000  x27 0000000000000000
02-12 04:16:07.658   505   505 F DEBUG   :     x28 0000000000000000  x29 0000007fc52b61f0
02-12 04:16:07.658   505   505 F DEBUG   :     lr  0000007f95516a9c  sp  0000007fc52b6150  pc  0000007f95516ac8  pst 0000000000001000
02-12 04:16:07.663   416   439 I ActivityManager: Collecting stacks for native pid 163
02-12 04:16:07.663   416   439 I system_server: libdebuggerd_client: started dumping process 163
02-12 04:16:07.664    50    50 I tombstoned: registered intercept for pid 163 and type kDebuggerdNativeBacktrace
02-12 04:16:07.664    50    50 I tombstoned: received crash request for pid 163
02-12 04:16:07.664    50    50 I tombstoned: enqueueing crash request for pid 163
02-12 04:16:07.668   505   505 F DEBUG   : 
02-12 04:16:07.668   505   505 F DEBUG   : backtrace:
02-12 04:16:07.668   505   505 F DEBUG   :       #00 pc 0000000000089ac8  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: e4cdaa3667d6b3fb35d46671d31909c3)
02-12 04:16:07.668   505   505 F DEBUG   :       #01 pc 00000000000062b0  /system/lib64/liblog.so (__android_log_default_aborter+12) (BuildId: b5b2ed9d2859e7c956635d8bdb685c6f)
02-12 04:16:07.668   505   505 F DEBUG   :       #02 pc 0000000000006e18  /system/lib64/liblog.so (__android_log_assert+336) (BuildId: b5b2ed9d2859e7c956635d8bdb685c6f)
02-12 04:16:07.668   505   505 F DEBUG   :       #03 pc 000000000018da9c  /system/lib64/libsurfaceflinger.so (android::renderengine::gl::GLESRenderEngine::create(android::renderengine::RenderEngineCreationArgs const&)+4320) (BuildId: d8310b0f3ed84e754d3d7031e2895e31)
02-12 04:16:07.668   505   505 F DEBUG   :       #04 pc 000000000018c86c  /system/lib64/libsurfaceflinger.so (android::renderengine::RenderEngine::create(android::renderengine::RenderEngineCreationArgs const&)+168) (BuildId: d8310b0f3ed84e754d3d7031e2895e31)
02-12 04:16:07.668   505   505 F DEBUG   :       #05 pc 000000000010c848  /system/lib64/libsurfaceflinger.so (android::SurfaceFlinger::init()+184) (BuildId: d8310b0f3ed84e754d3d7031e2895e31)
02-12 04:16:07.668   505   505 F DEBUG   :       #06 pc 00000000000021b8  /system/bin/surfaceflinger (main+368) (BuildId: 2a5c373dbea113a5035b20ca4aed30fe)
02-12 04:16:07.668   505   505 F DEBUG   :       #07 pc 000000000008506c  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+108) (BuildId: e4cdaa3667d6b3fb35d46671d31909c3)
```

develperbayman, 2023-02-12:
nice you will have the only known working distro on opi5 that waydroid actually works!

develperbayman, 2023-02-12:
but yeah for me it just hangs 

ubuntu@orange-pi:~$ waydroid show-full-ui
[00:17:54] Starting waydroid session
[gbinder] Service manager /dev/binder has appeared

then notta ...whats even weird is it looks like its running but??

[image]

*****logs*****
(001062) [00:17:21] % mount -o ro /var/lib/waydroid/images/system.img /var/lib/waydroid/rootfs
(001062) [00:17:21] % mount -t overlay -o ro,lowerdir=/var/lib/waydroid/overlay:/var/lib/waydroid/rootfs,upperdir=/var/lib/waydroid/overlay_rw/system,workdir=/var/lib/waydroid/overlay_work/system,xino=off overlay /var/lib/waydroid/rootfs
(001062) [00:17:21] % mount -o ro /var/lib/waydroid/images/vendor.img /var/lib/waydroid/rootfs/vendor
(001062) [00:17:21] % mount -t overlay -o ro,lowerdir=/var/lib/waydroid/overlay/vendor:/var/lib/waydroid/rootfs/vendor,upperdir=/var/lib/waydroid/overlay_rw/vendor,workdir=/var/lib/waydroid/overlay_work/vendor,xino=off overlay /var/lib/waydroid/rootfs/vendor
(001062) [00:17:21] % mount -o bind /var/lib/waydroid/waydroid.prop /var/lib/waydroid/rootfs/vendor/waydroid.prop
(001062) [00:17:21] Save config: /var/lib/waydroid/waydroid.cfg
(001062) [00:17:21] % lxc-start -P /var/lib/waydroid/lxc -F -n waydroid -- /init
(001062) [00:17:21] New background process: pid=33812, output=background
(001062) [00:17:21] % lxc-info -P /var/lib/waydroid/lxc -n waydroid -sH
lxc-start: waydroid: cgroups/cgroup2_devices.c: bpf_program_load_kernel: 332 Operation not permitted - Failed to load bpf program: (null)
RUNNING
(033747) [00:17:21] Failed to start Clipboard manager service, check logs
(034151) [00:17:46] Starting waydroid session
(034151) [00:17:46] org.freedesktop.DBus.Python.RuntimeError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 715, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/waydroid/tools/actions/container_manager.py", line 34, in Start
    do_start(self.args, session)
  File "/usr/lib/waydroid/tools/actions/container_manager.py", line 133, in do_start
    raise RuntimeError("Already tracking a session")
RuntimeError: Already tracking a session

(034151) [00:17:46] RuntimeError: Already tracking a session
(001062) [00:17:51] % lxc-info -P /var/lib/waydroid/lxc -n waydroid -sH
RUNNING
(001062) [00:17:51] % lxc-stop -P /var/lib/waydroid/lxc -n waydroid -k
lxc-start: waydroid: conf.c: run_buffer: 321 Script exited with status 126
lxc-start: waydroid: start.c: lxc_end: 988 Failed to run lxc.hook.post-stop for container "waydroid"
(001062) [00:17:52] % lxc-info -P /var/lib/waydroid/lxc -n waydroid -sH
STOPPED
(001062) [00:17:52] % /usr/lib/waydroid/data/scripts/waydroid-net.sh stop
vnic is waydroid0
(001062) [00:17:52] % systemctl is-enabled -q nfcd
Failed to get unit file state for nfcd.service: No such file or directory
(001062) [00:17:52] % umount /var/lib/waydroid/rootfs/vendor/waydroid.prop
(001062) [00:17:52] % umount /var/lib/waydroid/rootfs/vendor
(001062) [00:17:52] % umount /var/lib/waydroid/rootfs/vendor
(001062) [00:17:52] % umount /var/lib/waydroid/rootfs
(001062) [00:17:52] % umount /var/lib/waydroid/rootfs
(034208) [00:17:54] Starting waydroid session
(001062) [00:17:54] % /usr/lib/waydroid/data/scripts/waydroid-net.sh start
vnic is waydroid0
(001062) [00:17:55] % systemctl is-active -q nfcd
(001062) [00:17:55] % chmod 777 -R /dev/ashmem
(001062) [00:17:55] % chmod 777 -R /dev/sw_sync
(001062) [00:17:55] % chmod 777 -R /sys/kernel/debug/sync/sw_sync
(001062) [00:17:55] % chmod 777 -R /dev/dri
(001062) [00:17:55] % chmod 777 -R /dev/fb0
(001062) [00:17:55] % chmod 777 -R /dev/video-enc0
(001062) [00:17:55] % chmod 777 -R /dev/video-dec0
(001062) [00:17:55] % mv /var/lib/waydroid/config_session /var/lib/waydroid/lxc/waydroid
(001062) [00:17:55] % mount -o ro /var/lib/waydroid/images/system.img /var/lib/waydroid/rootfs
(001062) [00:17:55] % mount -t overlay -o ro,lowerdir=/var/lib/waydroid/overlay:/var/lib/waydroid/rootfs,upperdir=/var/lib/waydroid/overlay_rw/system,workdir=/var/lib/waydroid/overlay_work/system,xino=off overlay /var/lib/waydroid/rootfs
(001062) [00:17:55] % mount -o ro /var/lib/waydroid/images/vendor.img /var/lib/waydroid/rootfs/vendor
(001062) [00:17:55] % mount -t overlay -o ro,lowerdir=/var/lib/waydroid/overlay/vendor:/var/lib/waydroid/rootfs/vendor,upperdir=/var/lib/waydroid/overlay_rw/vendor,workdir=/var/lib/waydroid/overlay_work/vendor,xino=off overlay /var/lib/waydroid/rootfs/vendor
(001062) [00:17:55] % mount -o bind /var/lib/waydroid/waydroid.prop /var/lib/waydroid/rootfs/vendor/waydroid.prop
(001062) [00:17:55] Save config: /var/lib/waydroid/waydroid.cfg
(001062) [00:17:55] % lxc-start -P /var/lib/waydroid/lxc -F -n waydroid -- /init
(001062) [00:17:55] New background process: pid=34270, output=background
(001062) [00:17:55] % lxc-info -P /var/lib/waydroid/lxc -n waydroid -sH
lxc-start: waydroid: cgroups/cgroup2_devices.c: bpf_program_load_kernel: 332 Operation not permitted - Failed to load bpf program: (null)
RUNNING

develperbayman, 2023-02-12:
installed pyclip seems to stop the clipboard service error except in the logs still there

Joshua-Riek, 2023-02-12:
Hmm did you edit /var/lib/waydroid/waydroid_base.prop with the below settings?
```
ro.hardware.gralloc=default
ro.hardware.egl=swiftshader
```
Then restart the waydroid container?
```
sudo systemctl restart waydroid-container.service
```
Also, mesa is installed and compiled in a special way for hardware acceleration to work and can't be reinstalled without some work. It's something I would need to look deeper into.

closed 2023-02-18 by Joshua-Riek

awesomefroggy29, 2024-12-23:
@Joshua-Riek is there any way to enable hardware acceleration?