Joshua-Riek/ubuntu-rockchip #8: There is a serious bug for users who using Chinese/Japanese/Korean.

ironiris, 2023-02-09:
There is a serious bug for users who using Chinese/Japanese/Korean.
There is no problem in Ubuntu for PC, but in Ubuntu for Orange Pi 5,

when connected as a normal user(ubuntu), the input method setting does not work properly.
For Chinese/Japanese/Korean, input methods such as ibus and uim are used.

If I connect as a normal user(ubuntu), the settings cannot be finished and it does not work,
but it works only if you connect as root (change ubuntu's privileges to 0).

Please check.

Thank you.

Joshua-Riek, 2023-02-09:
Could you provide some detail on how I can reproduce this? Are you changing the input source in the keyboard settings, and it won't work properly when set to another language such as Korean?

ironiris, 2023-02-10:
After adding the Korean language to Ubuntu, an input method called iBus is automatically installed.
However, in the case of Armbian or Ubuntu for PC, iBus runs normally and the settings can be completed, but in the case of ubuntu-orange-pi5, can't be completed.
However, if uid/gid of the logging in user(ubuntu) is modified to 0/0, it is executed, but it cannot be regarded as a normal case.

Joshua-Riek, 2023-02-10:
I'm having trouble replicating this issue. I Install the Korean language from Settings -> Region & Language -> Manage Installed languages -> Install / Remove languages, then select Korean and install system-wide I get no errors. Then if I move to Settings -> Keyboard and add the Korean input source, I also get no errors. I then log the ubuntu user out and back in to see that the Korean language is working correctly.

What input settings are not working? How do I enable them to see the issue here? And what is the expected result?

[image]

ironiris, 2023-02-10:
The "ko" icon in the upper right corner is a normal state. I tried changing the settings, but I can't and it was strange, so I installed Ambian and there was no problem.
I'll try again.

you can close this issue.

closed 2023-02-10 by Joshua-Riek