Joshua-Riek/ubuntu-rockchip #13: xrdp server. login from windows10 got black screen after loginned

lowelowelowe, 2023-02-12:
xrdp server. login from windows10 got black screen after loginned

I tried many way to mess with the config files, notthing work

in /etc/xrdp/startwm.sh
added
export GNOME_SHELL_SESSION_MODE=ubuntu
export XDG_CURRENT_DESKTOP=ubuntu:GNOME
unset DBUS_SESSION_BUS_ADDRESS
unset XDG_RUNTIME_DIR

in /etc/gdm3/custom.conf
WaylandEnable=false

Joshua-Riek, 2023-02-12:
I recommend against using xrdp or any X application. It causes weird issues with the GPU and hardware acceleration, which is why I default the OS to use wayland. But this does raise an interesting topic of getting rdp to work with wayland. Ill need to look into this.

Joshua-Riek, 2023-02-12:
I just did a quick test using xfreerdp on Linux. It does connect and I can control the system with my mouse. But as you can see in the screenshot below there are some graphical issues, I will try on Windows 10 later today. I documented my steps below for future reference.

### Enable Remote Desktop
1. Open the settings application.
2. Click on Sharing to open the panel.
3. If the Sharing switch in the top-right of the window is set to off, click to switch it on.
4. Click Remote Desktop.
5. Set the Remote Desktop switch to on.
6. Set the Remote Desktop switch to on.
7. Check Enable Legacy VNC Protocol.

### Connect with rdp
1. xfreerdp /u:ubuntu /p:root /v:192.168.1.77 

### Screenshot
[image]

Joshua-Riek, 2023-02-12:
For Linux users, Remmina works as a VNC client while following the above instruction to enable remote desktop.
[image]

Joshua-Riek, 2023-02-12:
It works as well on Windows 10. Maybe when you were trying to set up RDP with xrdp server, some settings got messed up? Try to enable remote desktop through the settings menu. If that does not work, try a fresh install. Let me know if you encounter any more issues!
[image]

lowelowelowe, 2023-02-13:
> It works as well on Windows 10. Maybe when you were trying to set up RDP with xrdp server, some settings got messed up? Try to enable remote desktop through the settings menu. If that does not work, try a fresh install. Let me know if you encounter any more issues! [image]

confirm, it worked.  For fresh install without install xrdp. 
Enable "remote desktop" in setting->sharing will work fine.

cheers

closed 2023-02-13 by Joshua-Riek

iamryanchia, 2023-07-11:
Got black screen too.

setting.sharing can work, but it is not convenient as requiring login  first to use. My rock 5b works as a server, so no monitor is connected to it.

Please give a solution if possible, otherwise i will try VNC. Thank you!