Joshua-Riek/ubuntu-rockchip #15: I can't log in to YouTube with the Chromium web browser.

ironiris, 2023-02-13:
In version 1.3, I was able to log in to YouTube with the Chromium web browser, but in version 1.4, I cannot log in.

Joshua-Riek, 2023-02-13:
Oh, I see the issue. I added --test-type to the chromium start flags in /usr/share/applications/chromium-browser.desktop. Remove that flag and reboot your system, and it should work. But the --no-sandbox warning will appear again.

Joshua-Riek, 2023-02-13:
To fix for the moment, remove `--test-type` from the file `/usr/share/applications/chromium-browser.desktop` then reboot. 

This does bring back the --no-sandbox warning, but ill deal with that later. See commit 724fd949fcd772469cce54141281501847f0a202.

closed 2023-02-13 by Joshua-Riek

lowelowelowe, 2023-02-14:
In version 1.4 I did this

sudo apt update

sudo apt upgrade

wget http://ports.ubuntu.com/ubuntu-ports/pool/universe/c/chromium-browser/chromium-browser_85.0.4183.83-0ubuntu2.22.04.1_arm64.deb

sudo apt install ./chromium-browser_85.0.4183.83-0ubuntu2.22.04.1_arm64.deb

sudo apt install firefox

then chromium browser will able to login to youtube,gmail with out warning

Joshua-Riek, 2023-02-14:
Yes, that will work, but chromium will no longer be hardware accelerated and use the GPU properly.