Joshua-Riek/ubuntu-rockchip #17: Running off NVME

moshtin, 2023-02-15:
Hi, been trying to get this image onto my NVME. I have been using etcher to clone the SD card onto the NVME on other images I have been using but unfortunately when I have used this image it won't boot. Any suggestions? Thanks

Joshua-Riek, 2023-02-15:
Currently, I designed the OS to boot only from the SD card; next on my list for this OS is getting NVME boot to work correctly. I will update you when I make any progress.

moshtin, 2023-02-15:
Ah thank you, really appreciate it. Very impressed with what you have done

Joshua-Riek, 2023-02-15:
Thanks, I've been putting quite a lot of work into this project!
I ordered myself a 2230 NVME drive, and it should be here by tomorrow.

moshtin, 2023-02-15:
Excellent news thanks! Very quick question, bit of a novice here what is the best way to update your build once it iss on SD card / NVME. Will sudo apt update / upgrade suffice? Or downloading a new image. Just mindful I don't want to break anything by updating the wrong way

Joshua-Riek, 2023-02-15:
Sadly the best way to update is by downloading a new image and flashing it to your SD card or NVME. Using sudo apt update / upgrade will only update the installed packages, but not fetch any changes that I'm making for the OS to work properly.

Joshua-Riek, 2023-02-15:
Booting from an NVME drive is now working, and I have updated the [v1.5](https://github.com/Joshua-Riek/ubuntu-orange-pi5/releases/tag/v1.5) release to include the changes!