Joshua-Riek/ubuntu-rockchip #20: Spdif passthrough not available

Danibos, 2023-02-19:
Hi!

the passthrough options aren't available in pavucontrol (see example in the attachment). Launching VLC from terminal i can see the following error: vlcpulse audio output error: digital pass-through stream connection failure: Not supported

Please let me know if you want to test something or you need more infos 

Thanks!
[image]

Joshua-Riek, 2023-02-19:
Interesting; what command are you using to launch VLC from the terminal so I can reproduce it?

Danibos, 2023-02-19:
I start VLC from terminal, then after reproducing a media file with AC3 or DTS audio the error message appears in terminal

If can help, the only OS with SPDIF working is the official Orange Droid OS, tested with Kodi there, and the audio device it's named Audiotrack (IEC)

Joshua-Riek, 2023-02-19:
I think I found someone who could get AC3 and DTS audio working on the Armbian forums, ill look into this further. 
https://forum.armbian.com/topic/25957-guide-kodi-on-orange-pi-5-with-gpu-hardware-acceleration-and-hdmi-audio/#comment-158528

Joshua-Riek, 2023-02-19:
Can you provide me with a sample media file that uses AC3 or DTS audio? I'm still trying to produce the AC3 or DTS audio error message.

Danibos, 2023-02-19:
> Can you provide me with a sample media file that uses AC3 or DTS audio? I'm still trying to produce the AC3 or DTS audio error message.

sure, I've uploaded a file here: https://app.box.com/s/hceq6585o3mjvu8byadmnl4ajdawzd2f

Danibos, 2023-02-19:
> I think I found someone who could get AC3 and DTS audio working on the Armbian forums, ill look into this further. https://forum.armbian.com/topic/25957-guide-kodi-on-orange-pi-5-with-gpu-hardware-acceleration-and-hdmi-audio/#comment-158528

I've tested this one, the SPDIF options shows up + HDMI 5.1 output, but there are are a constant noises and finally is not working in any media player

i've tried to change some options as suggested here at the final of "readme", after compatibility: https://github.com/darealshinji/dcaenc but with not luck

Joshua-Riek, 2023-02-19:
Have you tried setting the output module to "ASLA output device"? The option can be found in the VLC ui under tools -> preferences -> audio. With this, I no longer see the digital pass-through error message.

Danibos, 2023-02-19:
I did, but not with your last release,  ill test it tomorrow and let you know
Thanks

Danibos, 2023-02-20:
Tested, unfortunately the issue persists
Tried several devices and to check un-check the SPDIF option
[image]

Joshua-Riek, 2023-02-22:
I noticed some playback issues with 4k videos in VLC, I'm going to try and work on that first, but hopefully we can come up with something!

Danibos, 2023-02-22:
Sure, no rush
Meanwhile I'm testing stuff

closed 2023-03-20 by Joshua-Riek