Joshua-Riek/ubuntu-rockchip #22: Boot from SATA m.2 SSD not NVME

Ninnix, 2023-02-21:
Hi, first of all thank you for the fantastic work you are doing and the support for this device.

i am trying to boot your ubuntu on SATA ssd NOT nvme. Simply writing the .img 1.6 ver the system boot but stuck here:

[image]

in general distros do not start from this type ssd directly at actual state of pi5, the process is well described [here](https://www.reddit.com/r/OrangePI/comments/10iq625/guide_how_to_boot_orange_pi_5_from_sata_m2_ssd/)

I looked inside /boot/ but there are no such files like /boot/orangepiEnv.txt or  /boot/ubuntuEnv.txt

are available for further tests.

Joshua-Riek, 2023-02-21:
Edit the file overlays.txt inside /boot/ to look like `fdt_overlays=rk3588-ssd-sata.dtbo`. I have not tested myself with a SATA SSD but this should work. I need to write up my own guide at some point

Ninnix, 2023-02-21:
Thanks you for the fast reply.

On sdcard installation I have add to /boot/firmware/overlays.txt `fdt_overlays=rk3588-ssd-sata.dtbo `

after that running `lsblk` I can see Sata SSD and partitions, so

So , I have mount ssd with an clean installation 

sudo mount /dev/sda1 /mnt/

sudo nano /mnt/overlays.txt and add `fdt_overlays=rk3588-ssd-sata.dtbo `

sudo umount /mnt/ && sudo sync

sudo poweroff

Remove SDcard and start again and it boot perfectly, thanks you, all seems okay, but I have random reboot using youtube on chromium, what can be the problem?

Joshua-Riek, 2023-02-21:
Awesome, glad it works! I have not seen the system reboot when using chromium, there might be a few graphical bugs, but nothing that should cause a system reboot though. Does the system always reboot when using youtube with chromium? Or was this a one-time occurrence?

Ninnix, 2023-02-21:
Every time same with twitch.tv, hw acceleration is working, but reboot after some, cooling and temp are okay

Joshua-Riek, 2023-02-21:
Very weird. Maybe the system can't get enough power so it's crashing? Try to remove the SSD and see if this also happens when just using the SD card.

Ninnix, 2023-02-21:
I have check now, and after 115 minutes bbut with two 1080p video rebooted also on microsd.

this is my charger, its not enough?

[image]

ferrellsl, 2023-02-21:
Orange Pi recommends a 5 volt/4 amp power supply.  The one you're using is too weak.

Joshua-Riek, 2023-02-21:
I just double-checked in the user manual, it does require a 5V/4A power supply.
[image]

Joshua-Riek, 2023-02-21:
I'll be closing this for now, if there are any issues or questions after you get the proper power supply please let me know.

closed 2023-02-21 by Joshua-Riek