Joshua-Riek/ubuntu-rockchip #23: feature request

develperbayman, 2023-02-22:
im struggling im trying to get this driver to work can you bake it into the next release?   they sold it to me at micro center in the pi isle sucks it dont work i keep getting errors on compile   
i would be eternally greatful

https://github.com/blahah/ralink_RT5370_linux_driver

ferrellsl, 2023-02-22:
Why don't you just buy a dongle that works?  The Ralink 5370 is over 12 years old and only supports 150Mbps 2.4Ghz.

This dongle below is only $3 USD and it also includes bluetooth, 600Mbps  dual band WiFi at 2.4/5.8Ghz and it works fine on my Orange Pi 5.

https://www.aliexpress.us/item/3256801435854396.html?spm=a2g0o.order_list.order_list_main.11.43a01802HBrdOa&gatewayAdapt=glo2usa&_randl_shipto=US

develperbayman, 2023-02-22:
do you think this one will work??

https://www.amazon.com/Bluetooth-1300Mbps-Wireless-External-Receiver-dp-B09KKLCHZZ/dp/B09KKLCHZZ/ref=dp_ob_title_ce

ferrellsl, 2023-02-22:
I doubt it since the ad fails to mention Linux.  It only mentions Windows and MacOS as being supported.  But one person responded that it does support Ubuntu, but that's probably x86_64 Ubuntu and not ARM based Ubuntu, so take his answer with a grain of salt.
https://www.amazon.com/ask/questions/Tx46FZU5ANIVNJ/ref=ask_ql_ql_al_hza

If we knew what chipset it used then we could know for sure, but the chipset isn't listed in the ad.
 

Message ID: ***@***.***>

develperbayman, 2023-02-22:
oh wow i didnt know i could do that with amazon ill prob just get the same one you originally posted i figured if i get a new dongle i wanna go all out though but i dont know how to verify itll work ??  

https://www.amazon.com/dp/B09XDTTG8Y/ref=sspa_dk_detail_1?psc=1&pd_rd_i=B09XDTTG8Y&pd_rd_w=aVsEs&content-id=amzn1.sym.dd2c6db7-6626-466d-bf04-9570e69a7df0&pf_rd_p=dd2c6db7-6626-466d-bf04-9570e69a7df0&pf_rd_r=6YAEA3EQSXHGAZ2N9F33&pd_rd_wg=46yES&pd_rd_r=63759d60-775f-4ba8-a497-26ee044266a4&s=electronics&sp_csd=d2lkZ2V0TmFtZT1zcF9kZXRhaWxfdGhlbWF0aWM&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEzRlpBVUIyOFVOVVNZJmVuY3J5cHRlZElkPUEwODMwMzUyMzM0MFFONTNNQzZUTSZlbmNyeXB0ZWRBZElkPUEwMDc5NDkzMjYyWUU2TUVOQ1FIViZ3aWRnZXROYW1lPXNwX2RldGFpbF90aGVtYXRpYyZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=

Joshua-Riek, 2023-02-23:
The drivers for the RT5370 are made for versions 3 and 4 of the Linux kernel, I doubt it will work without some modifications since the Orange Pi 5 uses the 5.10 kernel. I have been using the m.2 wifi module and have not tested any USB wifi devices. I know the adapter ferrellsl linked to at aliexpress works; I'm not sure about the amazon ones.

Also, I've seen code commits preparing for an Orange Pi 5b with built-in wifi support. It has not been announced but it is something to look out for in the future!

closed 2023-02-24 by Joshua-Riek