Joshua-Riek/ubuntu-rockchip #26: nothing saves on reboot

develperbayman, 2023-02-24:
install image then set up everything a sweet ....reboot ....wheres all my programs and settings? .....booting from sdcard  had this happen on both 1.5 and 1.6

Joshua-Riek, 2023-02-24:
That is very strange. I'll double-check on my end to see if I have the same issue with both versions, but maybe your sdcard is going bad? 

If you could provide some logs for me to look through on the system that did not save any settings I might be able to figure it out. You can save the dmesg and journalctl logs to files with the command below then upload them here for me to look.
```
dmesg > dmesg.log 
journalctl > journalctl.log
```

Joshua-Riek, 2023-02-24:
I just tested with both versions and it saved my settings and programs when i rebooted the system. My guess is your SD card is going bad? Or some setting your changing causes some weird issue.

develperbayman, 2023-02-24:
i figured it out basically i cannot have any other drive plugged in while booting because for some reason ....not just this distro but every distro for the orange pi has this feature ....its almost like it tries to use the drive as a system drive copies files over and then boots from sdcard then uses the drive idk .....im just using the other drives as storage .....now im running into sound issues something todo with installed packages  .....nothing to do with this project im sure  but its weird system sound works just no sound from any web browser not sure what to do i set all browsers page setting to allow unsecured content and sound ect ect ....google not really helping me i fear ....im off work for the weekend so i can play   ....is there a good tool your aware of that lets you easily manage sound system

Joshua-Riek, 2023-02-24:
I see that is very strange. As for sound Pauvcontrol is installed by default to manage the system sound, you should be able able to edit and see the volume control for each application using it.

develperbayman, 2023-02-24:
alright so i know what i did  ...basically i used a large markings file for synaptic package manager 15 gb worth of data im thinking it reset all my packages what im stumped on is... mv ~/.pulse ~/.pulse_backup ...is not working system sound works ...games work .....just not web ....this is defiantly weird

Joshua-Riek, 2023-02-24:
I do have some preinstalled packages that are held using `apt-mark hold`, If the synaptic package manager removed those held packages, things could easily break. But I don't think it would affect the browser sound.

develperbayman, 2023-02-24:
here in a bit ill post some extensive logs ....i have a amazon 10 port usb hub 1tb hard drive scsi hard drive and a 8 tb hard drive on the hub a 128 gb sdcard

Joshua-Riek, 2023-02-24:
Also, if you have a script or have written down the exact packages installed, I could try and reproduce it on my end.

develperbayman, 2023-02-24:
dpkg		install
kerneloops		install
openvpn		install
fontconfig		install
libvulkan1		install
libgdal30		install
libnetfilter-conntrack3		install
libqt5webengine-data		install
php-symfony-config		install
lsb-release		install
libevent-pthreads-2.1-7		install
gconf2-common		install
libsphinxbase3		install
python3-parso		install
openarena-oacmp1		install
redeclipse-server		install
cifs-utils		install
libfreeimage3		install
libssh2-1		install
libsynctex2		install
libperl5.34		install
openjdk-11-jdk		install
libsdl-mixer1.2		install
vlc-plugin-qt		install
libopenblas0		install
openjdk-11-jre		install
libblkid-dev		install
libcups2		install
libkms1		install
libopencolorio1v5		install
evtest		install
libxcb-present0		install
libvlc5		install
freedink-dfarc		install
libxapian30		install
libwrap0		install
python3-jedi		install
fonts-kacst-one		install
kodi-pvr-hdhomerun		install
libcanberra-gtk3-0		install
dnsmasq		install
libkeyutils1		install
containerd.io		install
python3-blinker		install
orca		install
qtchooser		install
debhelper		install
libkf5widgetsaddons5		install
mplayer-skins		install
libselinux1-dev		install
libxtables12		install
libcurl4		install
libpolkit-agent-1-0		install
libupower-glib3		install
libxapp1		install
u-boot-tools		install
sshpass		install
libgomp1		install
kodi-screensaver-pingpong		install
kodi-eventclients-dev-common		install
libtimedate-perl		install
libhtml-tagset-perl		install
initramfs-tools-core		install
python2.7-minimal		install
liblirc-client0		install
freeciv-client-qt		install
freedink		install
libieee1284-3		install
libpulse0		install
libwebpmux3		install
[... 2304 lines cut ...]
proj-data		install
texlive-latex-recommended		install
python3-savitar		install
libtbb12		install
libexo-common		install
libpugixml1v5		install
passwd		install
libkeybinder-3.0-0		install
libspectre1		install
libxcb-xfixes0		install
linux-libc-dev		install
upower		install
libproc-processtable-perl		install
libavfilter7		install
libnspr4		install
libgcc-s1		install
debconf-utils		install
wesnoth-1.16-server		install
libxcb-xv0		install
libevdev2		install
fonts-open-sans		install
libsratom-0-0		install
libgdk-pixbuf2.0-common		install
freeciv-sound-standard		install
php-symfony-filesystem		install

develperbayman, 2023-02-24:
`dpkg		install
kerneloops		install
openvpn		install
fontconfig		install
libvulkan1		install
libgdal30		install
libnetfilter-conntrack3		install
libqt5webengine-data		install
php-symfony-config		install
lsb-release		install
libevent-pthreads-2.1-7		install
gconf2-common		install
libsphinxbase3		install
python3-parso		install
openarena-oacmp1		install
redeclipse-server		install
cifs-utils		install
libfreeimage3		install
libssh2-1		install
libsynctex2		install
libperl5.34		install
openjdk-11-jdk		install
libsdl-mixer1.2		install
vlc-plugin-qt		install
libopenblas0		install
openjdk-11-jre		install
libblkid-dev		install
libcups2		install
libkms1		install
libopencolorio1v5		install
evtest		install
libxcb-present0		install
libvlc5		install
freedink-dfarc		install
libxapian30		install
libwrap0		install
python3-jedi		install
fonts-kacst-one		install
kodi-pvr-hdhomerun		install
libcanberra-gtk3-0		install
dnsmasq		install
libkeyutils1		install
containerd.io		install
python3-blinker		install
orca		install
qtchooser		install
debhelper		install
libkf5widgetsaddons5		install
mplayer-skins		install
libselinux1-dev		install
libxtables12		install
libcurl4		install
libpolkit-agent-1-0		install
libupower-glib3		install
libxapp1		install
u-boot-tools		install
sshpass		install
libgomp1		install
kodi-screensaver-pingpong		install
kodi-eventclients-dev-common		install
libtimedate-perl		install
libhtml-tagset-perl		install
initramfs-tools-core		install
python2.7-minimal		install
liblirc-client0		install
freeciv-client-qt		install
freedink		install
libieee1284-3		install
libpulse0		install
libwebpmux3		install
[... 2304 lines cut ...]
proj-data		install
texlive-latex-recommended		install
python3-savitar		install
libtbb12		install
libexo-common		install
libpugixml1v5		install
passwd		install
libkeybinder-3.0-0		install
libspectre1		install
libxcb-xfixes0		install
linux-libc-dev		install
upower		install
libproc-processtable-perl		install
libavfilter7		install
libnspr4		install
libgcc-s1		install
debconf-utils		install
wesnoth-1.16-server		install
libxcb-xv0		install
libevdev2		install
fonts-open-sans		install
libsratom-0-0		install
libgdk-pixbuf2.0-common		install
freeciv-sound-standard		install
php-symfony-filesystem		install`

develperbayman, 2023-02-24:
i tried to copy the file itself but its not letting me and the log files are doing it as well

develperbayman, 2023-02-25:
Ok so i got the bright idea i flashed my hard drive direct plugged it into my top usb port ..flatline ....is that the proper way?   ....i can open another rabbit hole for this topic?

Joshua-Riek, 2023-02-25:
Does your hard drive have an external power supply?

develperbayman, 2023-02-25:
No usb all the way

Joshua-Riek, 2023-02-25:
Ah, that is most likely the issue; you will need some external power source. The USB 3.0 specification can only power up to 5 watts which can be an issue with hard drives.

Joshua-Riek, 2023-02-25:
There are packages from the vendor-provided Orange Pi5 distro in the logs you provided, they can break this port of Ubuntu or cause unknown issues. Idk if these were installed manually or you might be on a different OS?

These are the packages that could cause conflict:
```
orangepi-firmware
linux-dtb-legacy-rockchip-rk3588
orangepi-zsh
orangepi-bsp-desktop-orangepi5
orangepi-config
balena-etcher-electron
orangepi-plymouth-theme
linux-image-legacy-rockchip-rk3588
linux-u-boot-orangepi5-legacy
orangepi-bsp-cli-orangepi5
orangepi-jammy-desktop-xfce
```

develperbayman, 2023-02-25:
so i reflased works great
this time around im going to grow this one organicly 

this plays morrowind flawlessly
still on sdcard

Joshua-Riek, 2023-02-25:
Awesome, ill close this issue for now, if you have any more issues, please let me know!

closed 2023-02-25 by Joshua-Riek