Joshua-Riek/ubuntu-rockchip #27: Can I install OBS studio?

ironiris, 2023-02-24:
I usually do YouTube streaming with my laptop and OBS software, but it would be great if I could do it on an SBC as well.

The 'apt install obs' command was installed without any errors, but when I try to run it, an error occurs.

Joshua-Riek, 2023-02-24:
Someone has ported OBS for this board [here](https://launchpad.net/~liujianfeng1994/+archive/ubuntu/rockchip-multimedia), but I don't use the same drivers, so the install could break hardware acceleration.

However, I have been working on a solution; I recently got Kodi up and running from the same ppa and will create a tutorial soon. Using the same method, I should be able to get OBS working too. I can test this out and let you know the results and how to set it up later today.  I'm glad you brought this up, I have been working on increasing video playback performance for the next release, it will be interesting to see how OBS performs.

ironiris, 2023-02-24:
Thank you for your hard work.

Joshua-Riek, 2023-02-24:
I'm going to convert this issue to a discussion.