Joshua-Riek/ubuntu-rockchip #29: Mouse pointer disappear after wake up on V1.6

ypatriot, 2023-02-24:
I got this while on wayland and x11

Joshua-Riek, 2023-02-24:
Are you using Ubuntu 20.04 or Ubuntu 22.04?

Joshua-Riek, 2023-02-24:
I see this occurs on Ubuntu 20.04, there are also a few other mouse bugs I have noticed. But I don't see this issue on Ubuntu 22.04. I've been thinking for a while now and will likely drop support of Ubuntu 20.04 in favor of Ubuntu 22.04; many drivers are not being backported for Ubuntu 20.04, or are in a broken state, making it difficult to maintain. If there is interest I could still maintain both, but the Ubuntu 20.04 image will be lacking many features, so I would prefer to focus on maintaining a single version of Ubuntu where I can make it the best I can.

Joshua-Riek, 2023-02-25:
I will close this for now unless this bug appears in Ubuntu 20.04.

ypatriot, 2023-02-25:
I'm using Ubuntu 22.04  v1.6 in nvme, mouse pointer is still sometimes disappear after wakeup
I also test Ubuntu 20.04 v1.7 on sdcars and it's the same issue

Joshua-Riek, 2023-02-25:
I see; I'll need to try and reproduce this on my side. Do you say it only happens sometimes? At the login screen or after already logged in the system?

Joshua-Riek, 2023-02-26:
I was able to reproduce this once, but I have yet to find a way to always produce it. I think I fixed this issue with this commit eee0b78efb6896c0818a11f7adb4b65ce9b11aa6. But I need to do some more testing

ypatriot, 2023-02-26:
is that patch file can be apply to the running orangepi5 ubuntu system? could you please give me guide to do ? thanks in advance

Joshua-Riek, 2023-02-26:
Of course! It's a small change and can be applied easily to a running system. Download and extract [this](https://github.com/Joshua-Riek/ubuntu-orange-pi5/files/10833937/rk3588s-orangepi-5.zip) archive; you will find the file `rk3588s-orangepi-5.dtb` inside.

Then transfer `rk3588s-orangepi-5.dtb` over to your running ubuntu system with a USB or however convenient.  You will need to open a command prompt and copy it to `/boot/firmware/rk3588s-orangepi-5.dtb` 

For example, say `rk3588s-orangepi-5.dtb` is on your desktop, and you open a new terminal there, you can copy it over with the below command:
```
sudo cp rk3588s-orangepi-5.dtb /boot/firmware/rk3588s-orangepi-5.dtb
```

After the file is copied, reboot your system, and the patch will be applied!

Joshua-Riek, 2023-02-28:
I have done more testing and have not run into this issue with the changes of eee0b78efb6896c0818a11f7adb4b65ce9b11aa6.

closed 2023-02-28 by Joshua-Riek

ypatriot, 2023-03-01:
Yes, agree, I test more than 10times screen blank, and the mouse cursor is available, thanks