Joshua-Riek/ubuntu-rockchip #30: ROS2 test

samuk, 2023-02-24:
Would someone with this hardware be willing to run this script to see if it installs? https://github.com/linorobot/ros2me

Joshua-Riek, 2023-02-24:
Of course, It seems to install without any issues. I'm unfamiliar with ROS 2 and its functionality, so I can't guarantee that It will fully work as this device is still relatively new.

samuk, 2023-02-25:
Great, thanks! If you wanted to try the next step then it would be to download this repo and run the script: https://github.com/linorobot/linorobot2  

Understand that you probably have other things to do though..

closed 2023-02-25 by samuk

Joshua-Riek, 2023-02-25:
No worries, I downloaded the next step, and it installed without issue! The issue I can see is if this uses any 3D rendering. The GPU drivers for the Orange Pi5 are still in the works; they are usable but have limitations.

samuk, 2023-02-25:
Nice! Thanks. ROS is primarily comms network anyway, So for most uses you'd do the 3D stuff on your laptop, the robot pi can be headless.