Joshua-Riek/ubuntu-rockchip #31: CM5 fork?

samuk, 2023-02-25:
I don't know if you'd be interested in maintaining a fork of this project for the upcoming [Radxa CM5? ](https://wiki.radxa.com/Rock5/CM5) It also uses the RK3588S.

I believe https://wiki.radxa.com/Special:SpecialContact are providing early access $free dev kits for those that request them & make a good case, in advance of the main release next month.

Joshua-Riek, 2023-02-25:
Interesting, Id love to maintain multiple boards, especially since it uses the same RK3588S, so drivers should be similar. But it also depends on the pricing of the hardware if I can't get my hands on a dev kit.

samuk, 2023-02-25:
Great, It's worth asking for a freebe!

closed 2023-02-25 by samuk