Joshua-Riek/ubuntu-rockchip #32: HW acceleration  break by connecting two monitors

Ninnix, 2023-02-25:
Connecting two monitors, one with hdmi and one with usb c the 'hardware acceleration breaks down. unplugging the monitor with usb c makes it work again.

Joshua-Riek, 2023-02-25:
Interesting, I have not yet tried using usbc for a monitor connection. I will try and see what I can find.

Joshua-Riek, 2023-02-25:
This is weird wayland seems to crash, and x11 is used. This will require some digging.

Ninnix, 2023-02-25:
Thanks you for you work and fast answer, if can I help in some Im here.

Joshua-Riek, 2023-02-25:
I have a usb c adapter that splits into two monitors, which works fine. 
Do you have one monitor plugged into the HDMI port and another into the usb c port?

Ninnix, 2023-02-25:
Yes its that

Joshua-Riek, 2023-02-26:
Fixed with commit eee0b78efb6896c0818a11f7adb4b65ce9b11aa6. Thanks for reporting this bug; I would never have found this myself!

closed 2023-02-26 by Joshua-Riek

Ninnix, 2023-03-09:
I have test today with 1.8 and still same

Joshua-Riek, 2023-03-09:
Weird. I will have another try with this.

Joshua-Riek, 2023-03-12:
Hey, I don't see any issue with the HW acceleration breaking in v1.8 and v1.9. 
Could you provide some more detail on what breaks and what you are doing?