Joshua-Riek/ubuntu-rockchip #33: Chromium "has trouble" playing this YouTube video

AndrewFischer, 2023-02-26:
I've been testing V1.7 for an hour -- I'm impressed. 

First problem I found is minor.  Chrome has trouble playing this YouTube video at 4K --> https://www.youtube.com/watch?v=bQSX04Hoxs0

Around 2:24 the view change to an overhead camera and Chromium falls apart.  Sometimes playback will freeze or drop to one frame every few seconds (key frames?)  Other times I'll get a white flash and then playback returns.    Occasionally Chromium will hang.  Takes many  tries for a crash.  

I realize the issue may be outside of this port.   Just  thought I'd should mention it in case it is it is an easy fix...

AndrewFischer, 2023-02-27:
I was already using that video to test two other systems.  It is easy to spot dropped frames. 

I will look and see if I can find any more videos where it fails.

Joshua-Riek, 2023-03-05:
I created a discussion post #52 about issues regarding Chromium.

closed 2023-03-05 by Joshua-Riek