Joshua-Riek/ubuntu-rockchip #35: Vncserver,  Qbittorrent, telegram, firefox

tarakula, 2023-02-27:
Vncserver Problem
i am using tightvncserver, when connecting to orangepi 5 using smartphone. it keep stuck on connecting screen

then I can't install several applications such as telegram, firefox, and qbittorrent via the terminal. On the terminal displays
`Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is heWaiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 7716 (apt) `

Joshua-Riek, 2023-02-27:
It looks like another apt instance is running on your system; either reboot your system or kill the apt process `sudo kill -9 7716`. You should be able to install your applications.

As for the vncserver issue, I helped another person with this but connecting over a windows 10 or Linux desktop. You can follow the instructions [here](https://github.com/Joshua-Riek/ubuntu-orange-pi5/issues/13#issuecomment-1427094343) to enable the remote desktop inside Ubuntu. I don't know of any applications that can be used on a smartphone.

tarakula, 2023-02-27:
thank you.

for vncserver alternative, i already try using nomachine. it work fine on android apps and windows

Joshua-Riek, 2023-02-27:
Awesome! If you have any other issues, please let me know!

closed 2023-02-27 by Joshua-Riek

tarakula, 2023-02-27:
how to install balena etcher? original imange orangepi 5 have it

Joshua-Riek, 2023-02-27:
Download the file `balena-etcher-electron_1.7.9+5945ab1f_arm64.deb` from [here](https://github.com/futurejones/balena-etcher-arm/releases/tag/v1.7.9). 

Then you can install with the below commands:
```
sudo apt-get update
sudo apt-get install -y gconf2 gconf-service libgconf-2-4 gconf-service-backend gconf2-common
sudo dpkg -i balena-etcher-electron_1.7.9+5945ab1f_arm64.deb
```