Joshua-Riek/ubuntu-rockchip #36: start ubuntu on nvme

adrielaquino, 2023-02-27:
I'm a newbie, I'm using ubuntu on the ssd card but I would like to use nvme, can I migrate directly from ssd to nvme?

Joshua-Riek, 2023-02-27:
Booting directly from a USB, NVMe, or SATA device requires flashing the bootloader to the SPI. I recommend downloading etcher and the latest release, then writing to your SSD if you wish to migrate I have included the instructions below. 

## If using a NVMe SSD

Flash the bootloader with the below command:
```
sudo dd if=/usr/share/orangepi/rkspi_loader.img of=/dev/mtdblock0
```

Then migrate data to your NVMe SSD (this can take a long time):
```
sudo dd if=/dev/mmcblk1 of=/dev/nvme0n1 bs=4k
```

## If using a SATA SSD

Flash the bootloader with the below commands:
```
sudo dd if=/usr/share/orangepi/rkspi_loader.img of=/dev/mtdblock0
echo "fdt_overlays=rk3588-ssd-sata.dtbo" | sudo tee /boot/firmware/overlays.txt
```

Migrate data to your SATA SSD (this can take a long time):
```
sudo dd if=/dev/mmcblk1 of=/dev/sda bs=4k
```

Joshua-Riek, 2023-02-27:
Also, if you would like to migrate your data using etcher you can download it to your system as below:

Download the file `balena-etcher-electron_1.7.9+5945ab1f_arm64.deb` from [here](https://github.com/futurejones/balena-etcher-arm/releases/tag/v1.7.9). 

Then you can install with the below commands:
```
sudo apt-get update
sudo apt-get install -y gconf2 gconf-service libgconf-2-4 gconf-service-backend gconf2-common
sudo dpkg -i balena-etcher-electron_1.7.9+5945ab1f_arm64.deb
```

adrielaquino, 2023-02-27:
worked, thank you.

closed 2023-02-28 by Joshua-Riek