Joshua-Riek/ubuntu-rockchip #37: Monitor Brightness Control

ferrellsl, 2023-02-28:
I'm unable to set my monitor brightness under both releases of v1.6 and v1.7.  Moving the brightness slider has no effect.

Joshua-Riek, 2023-02-28:
The brightness slider won't adjust a monitor's brightness; it's meant for an LCD. 

Ubuntu adds that brightness slider if any devices are found under `/sys/class/backlight/`, typically LCDs. The Orange Pi vendor has two devices defined under this path by default, I can only assume it's for some LCD attachments.

I can remove this, but I think it would cause issues if someone tried to use an LCD. I'll see if I can have the UI disable that brightness setting, as it can cause some confusion. Or I might need to create a device tree overlay, ill need to do some further research on this.

ferrellsl, 2023-02-28:
I'm using an LCD panel and the slider doesn't work.  I'm not using a CRT.

ferrellsl, 2023-02-28:
Here's a link to the panel:  [https://tinyurl.com/3k32k3wf](url)

ferrellsl, 2023-02-28:
I've tried 2 LCD panels from different manufacturers to rule out bad cabling or non-standard panels.  The other panel is a Hewlett Packard 2210m and the slider has no effect with either LCD panel.  I'm using the HDMI ports on both panels, not USB C.

Joshua-Riek, 2023-02-28:
I meant an LCD that connects over a ribbon cable, not HDMI or USB C. Like this from Orange Pi: http://www.orangepi.org/html/hardWare/otherAccessories/details/Orange-Pi-Touch-Screen-Pi5.html

ferrellsl, 2023-02-28:
Ah, OK.  That makes sense.

Joshua-Riek, 2023-03-16:
I finally found the Orange Pi 5 Touch Screen LCD on AliExpress. Once it arrives, I can try to remove the brightness slider. I will need to have it only enabled when this type of LCD is connected.

closed 2023-10-28 by Joshua-Riek