Joshua-Riek/ubuntu-rockchip #38: Bluetooth Issue on Realtek 8821CU

ferrellsl, 2023-02-28:
I'm using a USB WiFi/Bluetooth combo dongle.  The WiFi chip is a Realtek 8821CU and works fine.  Bluetooth works, but only after unplugging and replugging the dongle. 

The vendor ID for the Bluetooth is USB\VID_0BDA&PID_C820&REV_0200&MI_00 which is a generic Realtek bluetooth device.

After some searching, I found this very recent thread for implementing a fix:  [https://ubuntuforums.org/showthread.php?t=2482906](url)
Would it be possible to roll this into the current or next release?  Thanks for you hard work.  I know you have to be extremely busy.

Joshua-Riek, 2023-02-28:
Sure, i can add this when I have some time! Could you give me a link to the USB WiFi/Bluetooth device or the exact model you are using? I would just like to make sure I can reproduce this on my hardware for testing!

ferrellsl, 2023-02-28:
Sure, here's the listing on AliExpress:  [https://tinyurl.com/2jbtvbhc](url)

Joshua-Riek, 2023-02-28:
I have the Orange Pi5 PCIe WiFi 6.0 module (AP6275P) enabled by default, I wonder if that could be causing a conflict.
Could you try the below commands on your system to disable the device?

```
sudo systemctl disable ap6275p-bluetooth
sudo systemctl stop ap6275p-bluetooth
echo "fdt_overlays=" | sudo tee /boot/firmware/overlays.txt
sudo reboot now
```

ferrellsl, 2023-02-28:
> I have the Orange Pi5 PCIe WiFi 6.0 module (AP6275P) enabled by default, I wonder if that could be causing a conflict. Could you try the below commands on your system to disable the device?
>
> ``` [...]

It had no effect.  I still have to unplug/replug the dongle before bluetooth can be enabled.  Not a dealbreaker, just a minor nuisance.

Joshua-Riek, 2023-02-28:
Interesting, I see that the kernel has a config option for the rtl8821cu. Could you try the command `sudo modprobe 8821cu` before plugging in the dongle?

I don't have the dongle but can see the rtl8821cu driver load after issuing the above command.
```
root@orange-pi5:~# sudo modprobe 8821cu
root@orange-pi5:~# dmesg | grep -i 8821cu
[  138.482839] usbcore: registered new interface driver rtl8821cu
root@orange-pi5:~# lsmod | grep -i 8821cu
8821cu               1843200  0
```

Joshua-Riek, 2023-02-28:
That is so strange. I'll need to wait until I have the dongle to debug correctly.

ferrellsl, 2023-02-28:
> That is so strange. I'll need to wait until I have the dongle to debug correctly.

OK.  No worries.  At least I have a workaround for the issue.  Sort of off-topic, but this dongle doesn't work at all with Orange Pi 5 Android.  Do you know of a USB dongle that does by any chance?

ferrellsl, 2023-02-28:
> > That is so strange. I'll need to wait until I have the dongle to debug correctly.
> 
> OK. No worries. At least I have a workaround for the issue. Sort of off-topic, but this dongle doesn't work at all with Orange Pi 5 Android. Do you know of a USB dongle that does by any chance?

Oops.  I was wrong about this adapter not working under Android.  It works if boot Android, insert the dongle and then under WiFi settings tell Android to enable WiFi automatically.

Ninnix, 2023-03-01:
Hi,

I also notice the same behaviour with this dongle. i often have to unplug and plug it in again on reboot, i have also tried it on all usb ports. But BT in general works fine, on the other hand nothing to do with wifi which is detected, detects networks but the connection always fails after entering the password 

[image]

Joshua-Riek, 2023-03-01:
I got the dongle and noticed another issue. When inserted into the blue USB ports, the system will delay the boot process for about 4 minutes trying to reset the USB controller, but this issue is beyond me. I also tested that this happens on the official orangepi os. Inserting the dongle into the white USB 2.0 port will not cause the boot delay.

I have a workaround for replugging the dongle after a reboot. Enter the below commands on your system: 
```
echo "rfcomm" | sudo tee -a /etc/modules
echo "8821cu" | sudo tee -a /etc/modules
echo "rtk_btusb" | sudo tee -a /etc/modules
```

ferrellsl, 2023-03-01:
> Hi,
> 
> I also notice the same behaviour with this dongle. i often have to unplug and plug it in again on reboot, i have also tried it on all usb ports. But BT in general works fine, on the other hand nothing to do with wifi (AC600) which is detected, detects networks but the connection always fails after entering the password
> 
> [image]

Yeah, that's really odd. WiFi always works flawlessly for me and it's the bluetooth that fails to initialize until I unplug/replug the adapter.  Once I do that, bluetooth works just fine.  I'm going to look for another 2-in-1 adapter and see if that solves my problem.  Gonna order this one and see how it performs:  https://www.amazon.com/EDUP-Bluetooth-1200Mbps-Receiver-Transmitter/dp/B099K2FD1J/ref=sr_1_11?keywords=usb%2Bwifi%2Badapter%2Bwith%2Bbluetooth&sr=8-11&th=1

ferrellsl, 2023-03-01:
> I got the dongle and noticed another issue. When inserted into the blue USB ports, the system will delay the boot process for about 4 minutes trying to reset the USB controller, but this issue is [...]

Wow, that's crazy.  I use my dongle in the blue ports and I don't experience the boot delay you're describing.  I wonder if there's a hardware quality control issue with these super inexpensive 2-in-1 adapters. I paid $3 USD for mine and as the saying goes, you get what you pay for.  Gonna try this one for $17 USD and see what happens.  I'll let you know how it works out once I receive it.  https://www.amazon.com/EDUP-Bluetooth-1200Mbps-Receiver-Transmitter/dp/B099K2FD1J/ref=sr_1_11?keywords=usb%2Bwifi%2Badapter%2Bwith%2Bbluetooth&sr=8-11&th=1

Joshua-Riek, 2023-03-01:
If anyone experiences a very long boot delay, it could be because of something in the blue USB ports. I don't know if anyone else has had this happen to them. 

@Ninnix let me know if your still having issues with connecting to wifi after issuing those commands and rebooting your system.

Ninnix, 2023-03-01:
> If anyone experiences a very long boot delay, it could be because of something in the blue USB ports. I don't know if anyone else has had this happen to them.
> 
> @Ninnix let me know if your still having issues with connecting to wifi after issuing those commands and rebooting your system.

I need to change something in command, no? Do you suggest to delete the first row in overlay.txt?

Joshua-Riek, 2023-03-01:
> I need to change something in command, no? Do you suggest to delete the first row in overlay.txt?

If you open the terminal and type the commands below, reboot your pi5, which should work. 
I'm wondering if your wifi can connect properly after this.
```
echo "rfcomm" | sudo tee -a /etc/modules
echo "8821cu" | sudo tee -a /etc/modules
echo "rtk_btusb" | sudo tee -a /etc/modules
```

ferrellsl, 2023-03-02:
The EDUP USB 2-in1 adapter that ordered arrived this morning (link in earlier post).  WiFi and Bluetooth work just fine without any issues.  I recommend that folks pay a few extra dollars for the EDUP adapter to avoid the issues we've seen with the Realtek 8821CU adapters.  The EDUP adapter sales info and the front of the box list the data rate at 1200Mbps which is twice that of the Realtek 8821CU adapter, but looking at the back of the box the data rate is listed as 866Mbps.  The price is $17 USD (free shipping) which won't break anyone's bank account but that IS twice what I paid for my Realtek 8821CU adapter when adding in the shipping costs from AliExpress.

ferrellsl, 2023-03-02:
Poked around and found that the EDUP adapter is based on the Realtek 88x2bu chipset.

Joshua-Riek, 2023-03-02:
I'm glad that the EDUP adapter works, I will have to try it myself. But there is not much I can do about the issues with the 8821CU, besides modifying the file /etc/modules to load the kernel modules for the dongle on startup.

closed 2023-03-03 by Joshua-Riek