Joshua-Riek/ubuntu-rockchip #39: RTC time not work Perfectly

ypatriot, 2023-02-28:
Helo, 
I report the new finding about RTC, actually I realized it has happened since Ubuntu22.04 V1.5, now I'm using V1.7.
here is the detail : 
I power-on at 6:00 (28/Feb/2023) the clock is update by internet time, then I shutdown the board in 8:00, and then I power on again in 16:30 without internet connection, the board clock is 1:00 and the date still the same  (28/Feb/2023) 
This is very important that the board clock always update while no internet connection, hope this will got your attention, Thank

Joshua-Riek, 2023-02-28:
Thanks for the find! You can sync the current time and date to the RTC with the below command:
```
sudo hwclock -w
```

Joshua-Riek, 2023-02-28:
I created a systemd service to synchronize the current time and date to hardware RTC automatically 692056c25dccebbdcf294d68247c413715935791.
Thanks for finding this issue! The fix will be included in the next release.

closed 2023-02-28 by Joshua-Riek

ypatriot, 2023-03-01:
Thanks a lot