Joshua-Riek/ubuntu-rockchip #40: how to use wiringpi?

KingingWang, 2023-03-01:
apt install wiringpi

and

gpio readall
Oops: Unable to determine board revision from /proc/device-tree/system/linux,revision
or from /proc/cpuinfo
 -> No "Hardware" line
 ->  You'd best google the error to find out why.

KingingWang, 2023-03-01:
and I compile the code

gpio readall
Oops: Unable to determine board revision from /etc/orangepi-release or /etc/armbian-release.
 -> Unable to open /etc/orangepi-release or /etc/armbian-release.
 ->  You'd best google the error to find out why.

Joshua-Riek, 2023-03-01:
Enter the below command, and it should work:
```
echo "BOARD=orangepi5" | sudo tee /dev/orangepi-release
```

Joshua-Riek, 2023-03-01:
I have been meaning to add wiringpi by default and will be doing so in the next release.

Joshua-Riek, 2023-03-02:
I added wiring pi in the commit 32207867f69b5e9b8ea3b5574c17b4e54483999f, which will be in the next release.

closed 2023-03-02 by Joshua-Riek