Joshua-Riek/ubuntu-rockchip #46: Chromium Unresponsive

ferrellsl, 2023-03-05:
When I try to use either Gmail or Yahoo Mail in Chromium and add attachments to an email that I want to send, the interface becomes totally unresponsive after the attachment gets uploaded to the server and I have to kill Chromium.  FireFox works without any issues in this regard.

Joshua-Riek, 2023-03-05:
I can only do limited things for Chromium since I'm not a Chromium developer, but I have been working on updating Chromium to a new version. Sadly all of the versions I test have way more glitches and bugs than the current. However, I still have a few other versions to test, maybe one of them will be more stable.

ferrellsl, 2023-03-05:
No worries.  FireFox is working fine so I'll stick to that for now.  I expect hiccups on newly released hardware.

Bluesmanuk, 2023-03-05:
Although related in a more tenuous manner, I find that if turning the onscreen keyboard on, it cannot be used with Chromium as it never popups up when selecting any of the entry fields for searches etc

Joshua-Riek, 2023-03-05:
I will convert this into a discussion, I currently have Chromium 91.0.4472.164 included in the Ubuntu images by default. I found this version from the Orange Pi vender, and I have no idea who maintains it. 

However, I know of another version, Chromium 110.0.5481.4, available [here](https://launchpad.net/~liujianfeng1994/+archive/ubuntu/rockchip-multimedia/+packages). From my testing, it has a lot of graphical issues and bugs, version 91.0.4472.164 is much more stable in its current state. I don't know who to reach out to for issues regarding Chromium.