Joshua-Riek/ubuntu-rockchip #49: USB boot does not work for me

Bluesmanuk, 2023-03-05:
I noted in the notes for v1.6 that USB booting had been enabled but having tried both this and v1.7 and all 4 USB ports, I was unable to get it to work.

Joshua-Riek, 2023-03-05:
USB boot should be working fine. First, download the latest v1.8 release and write it to your SD Card, then boot into the system and enter the below command:
```
sudo dd if=/usr/share/orangepi/rkspi_loader.img of=/dev/mtdblock0
```
The above command can take up to 5 minutes to finish. But after running it, you can boot from a USB device. But be sure to use the top blue USB port; the others are USB 2.0 and can sometimes cause issues.

Bluesmanuk, 2023-03-05:
Thanks.

I managed to get it to work after entering the commend twice. It's great to be able to use this option.

Joshua-Riek, 2023-03-05:
I'm glad it's working now; I plan to create a script with some error checking to ensure no issues occur.

closed 2023-03-05 by Joshua-Riek