Joshua-Riek/ubuntu-rockchip #50: Unable to install newer versions of Kodi

Bluesmanuk, 2023-03-05:
I was keen to see if Kodi would be able to play videos with hardware acceleration and followed the Kodi wiki to install a stable latest version and also latest nightly dev builds but whilst the process seems to complete, no app icon is created in order to run it.

I am only able to install an old 19.4 release that has no hardware accelerated playback.

Joshua-Riek, 2023-03-05:
I have been able to get Kodi working with hardware acceleration, I need to document the process as it requires a few commands to be entered, and specific settings must be configured.

Joshua-Riek, 2023-03-05:
I created a Wiki page on how to install Kodi properly. I do not maintain Kodi itself and the application still has issues, but it should work nonetheless.
https://github.com/Joshua-Riek/ubuntu-orange-pi5/wiki/Kodi

Bluesmanuk, 2023-03-05:
Thanks.

Following the details I managed to get it working (it installs V19.4) but it does present some odd results.

All HDR videos show thumbnails of solid colour green and when played only shows a black screen.
When pressing the O button to get hardware acceleration, I first found that a video that was playing with black bars top and bottom, only the fist couple of lines of the normal output were being shown, one of which did confirm that acceleration was working But then I played another video and it seemed that the video screen was sitting atop of the underlying interface, with the usual controls for subtitle, video/audio settings etc not being available.
But then I played some videos which normally occupy the whole screen and none of the controls were available and the O button showed nothing on screen. To get out of playback I had to continually ship forward until the video completed playback and returned to the main Kodi interface.

I know that Kodi is not yet optimised for this new processor but it was strange behaviour that  have not seen before.

One other thing of note is that logging in to run Kodi in this manner prevents me from connecting to my bluetooth headphones because the Kodi interface does not natively support this function.

I'll keep testing new versions and reporting my findings.

Joshua-Riek, 2023-03-05:
HDR does not work, it needs to be implemented in Wayland, currently being worked on but i have no idea when it will properly work https://gitlab.freedesktop.org/wayland/weston/-/issues/467.

As for the Kodi issues i noted these too, but it seems to be a problem with Kodi, I don't really know who to contact about that. However I'm going to convert this into a discussion post so people can post their findings.