Joshua-Riek/ubuntu-rockchip #57: freezing on bootup

davidstoll, 2023-03-07:
First of all, thank you for making this build.  It's awesome!

I have been playing with it and see what you mean about uninstalling anything.  :)

I have tried to install a few things via command line.  Should I use an appstore or is command line better for now.  It's just nice to be able to browse.  If an app store is ok, which do you suggest.

Lastly, I have installed a few things, again, just playing around a bit and I have now twice gotten to the point where the bootup freezes and won't allow me to log in.  I'm pretty sure it was a program that I installed.  So, my question is, how can I just boot to a command prompt so I can apt remove the program and see if that helps?  Previously, I just blew it away and started fresh, but now that it has happened a second time (with a different program), I'm wondering if there is a way to avoid a total redo.

Thanks again!

Joshua-Riek, 2023-03-07:
Hmmm. This is strange. Could you provide me some more detail on the system freezing on bootup? Like do you see the splash screen? Is it blank? Does the monitor turn on at all?

Most boot-related issues are power related, do you know what your power supply is rated for? Orange PI5 requires a 5V/4A power supply. Anything lower than that could cause issues, especially during the boot process.

davidstoll, 2023-03-07:
I have the 4A power plug that it came with.  I also have a spare, which I just tried and it doesn't matter.  It's clearly me messing around installing things.

So, the first install froze at the login screen.  The mouse and USB keyboard would not respond, but all the splash screens came up just fine.  The second time, I set the GUI to auto-login in settings.  I also logged into Ubuntu One.  So, it now freezes at the point where I can see the jellyfish background thing, but it's asking me for my root login because it's saying something about my keyring.  But the keyboard and mouse do not respond.  if I wait long enough...like 10+ minutes, it times out BACK to the black login screen (even though it should be logging in automatically, which it did successfully before I started mucking around), the keyboard and mouse wake up so I can type in the password, hit enter and then once again, I see the jellyfish thing and top panel with the date/time, ethernet and sound icons, but I can't click anything.  No favorites bar or launch menu for a few minutes and then eventually the favorites bar comes up.  But I can't click anything or use the keyboard shortcuts or anything.  It's like the CPU is crunching on something which is actually making it react very slowly.

I was able to do ctrl alt F# to get to a prompt and remove the last program I installed, but that doesn't seem to help.

motorics, 2023-03-07:
> Here are the two logs. I'm going to use the 1.8 that I see is up now and
> overwrite the 1.7 in a few minutes unless you don't want me to. I think I
> downloaded it just before 1.8 went up. Thanks again for all of this work!
>
> dpkg.log
> b3b010fd25424c3aab7c173cf71fdfd9.zip

When it is freezed there is just a prompt bar blinking in the upper left
corner of the screen? I know similar behavior when your SD card is full.

Thank you!

davidstoll, 2023-03-07:
No, it's the normal ubuntu login.

davidstoll, 2023-03-07:
Or just the Ubuntu desktop GUI.

davidstoll, 2023-03-07:
Sorry....and the SD card has dozens of gigs available.

Joshua-Riek, 2023-03-08:
Let me know if this issue happens on v1.8. Its very weird, i don't see anything out of place in the logs.

Joshua-Riek, 2023-03-18:
I have not been able to reproduce this. If there are still any issues, please let me know.

closed 2023-03-18 by Joshua-Riek