Joshua-Riek/ubuntu-rockchip #58: GPU rasterization not on in Chromium by default

davidstoll, 2023-03-07:
I noticed that "GPU rasterization" could be enabled in the flags option of Chromium.  Once I do that, then all but Vulcan are enabled and green.

Is there any way you could turn that on by default?  Or maybe there is a reason to keep it off?

Thanks!

Joshua-Riek, 2023-03-07:
I could turn it on by default quite quickly. But I would be cautious, I'm not sure if it would cause issues, so I would need to look more into it as the current GPU drivers are still new and have quite a lot of bugs.

One quick example is the --no-sandbox message that pops up. If I disable the flag option for --no-sandbox Chromium GPU acceleration would break.

diepquynh, 2023-03-09:
Does that --no-sandbox flag does something magically? Because I tried to run without it, things went fine for some time and then my OPi5 does some random freeze (the screen basically freezes, but my keyboard still responds to caps lock key somehow)

Tested on v1.8 image

Joshua-Riek, 2023-03-09:
If I recall correctly that flag is required for the GPU to work with this version of Chromium and can cause issues when not enabled. 
I will try removing the --no-sandbox flag and see what happens, it's been a while since I tested to see what happens. But it's strange your system froze, I wonder if that flag causes some sort of memory leak if not used.

diepquynh, 2023-03-09:
Here's what happen when I run Chromium 110 with the same flags as Chromium 91 (which was shipped with v1.8 image)
It occasionally happens, but most of the time when the screen starts freezing, it looks like this
[image]

Joshua-Riek, 2023-03-09:
Yeah, I have been working on upgrading the GPU drivers and Chromium version to 110. I currently use a custom version of panfork and the x11 blob so it can cause some issues with that new version of Chromium.

When using the mesa drivers from amazingfate's ppa, I have not seen any weird visual artifacts with Chromium 110. But the performance and glitches in the browser seem worse than in version 91.

Joshua-Riek, 2023-03-10:
I updated Chromium in the latest release to version 110. There should not be any weird visual artifacts.
[image]

diepquynh, 2023-03-11:
On v1.9, no issues with visual artifacts so far. However, there happens to have weird texts and component colors on several webpages like Shopee or YT (below are one of the examples, there are a lot more than that)

This issue is not there on Chromium 91. I wonder if there are changes related to Chromium renderer itself 
[image]
[image]

diepquynh, 2023-03-11:
Looks like Chromium 91 was shipped with some sort of magic to work with Mali driver
[image]

Joshua-Riek, 2023-03-11:
Chromium 91 is made for x11 and to be used with the blob x11 driver [here](https://github.com/Joshua-Riek/ubuntu-orange-pi5/files/10948833/libmali-valhall-g610-g6p0-x11-gbm_1.9-1_arm64.zip). I removed it in the latest update for more compatibility using wayland.

closed 2023-03-12 by Joshua-Riek