Joshua-Riek/ubuntu-rockchip #59: PCIe WiFi 6.0 Module (AP6275P)

davidstoll, 2023-03-07:
I was pointed to this on another page, but it doesn't fit. oops  https://www.amazon.com/gp/product/B08NSSJNV1

Do you know where I can get the official module that you reference in your FAQ?  I think Amazon is just out of stock.

Also, sorry for making all of these posts, but I figured they would be better to be separate.

Ninnix, 2023-03-07:
https://github.com/Joshua-Riek/ubuntu-orange-pi5/wiki/PCIe-WiFi-6.0-Module-(AP6275P)

davidstoll, 2023-03-07:
This is the info, but do you have a source for purchasing it?

Joshua-Riek, 2023-03-08:
I purchased the WiFi Module from [this](https://www.amazon.com/gp/product/B0BRSLNT5W/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) page, but it seems to be out of stock for now.

davidstoll, 2023-03-08:
Thanks, I'll keep an eye on that.

By the way, did you know this guy reviewed your work?  I follow him and
like what he does.  That's how I found you.  Your build is the best I've
tried.

https://youtu.be/kgcyKUtUqQs

Thanks again!

Joshua-Riek, 2023-03-08:
I'm looking around and don't see any for sale. I recommend [this](https://www.amazon.com/dp/B099K2FD1J?psc=1&ref=ppx_yo2ov_dt_b_product_details) dongle if you need one. 
Yeah, seeing people enjoying my work is fantastic. All the feedback has been a big help too, it's hard to hunt down bugs myself.

closed 2023-03-08 by Joshua-Riek

davidstoll, 2023-03-11:
Would this work?
https://www.newegg.com/p/0XM-015W-00007

Joshua-Riek, 2023-03-11:
I don't know, but I found this post about some working wifi adapters: https://jamesachambers.com/using-m2-wifi-adapters-with-orange-pi-5/