Joshua-Riek/ubuntu-rockchip #61: No cpu.weight

adubkov, 2023-03-09:
Not sure it is a kernel issue or cpu doesn't support this feature, but don't see `cpu.weight` in `/sys/fs/cgroup/`.
Without it I can't run microk8s which was whole point of getting orange pi :(

adubkov, 2023-03-09:
Or maybe it is something else, just keep failing with `invalid capacity 0 on image filesystem`

adubkov, 2023-03-09:
I will reopen it as it seems to be related to the kernel build:

https://stackoverflow.com/questions/71739599/cgroup-not-found-the-cpu-weight-in-sys-fs-cgroup

```
PDS CPU scheduler doesn't support all the features CFS does. If you rebuild with CFS instead of PDS, you'll get the expected /sys/fs/cgroup/cpu.weight.
```

Joshua-Riek, 2023-03-09:
I will need to look into the available kernel configuration options for this, I'm not familiar with cpu.weight.

adubkov, 2023-03-09:
Have you tried run k8s on it? I'm experiencing it when running microk8s.

Joshua-Riek, 2023-03-09:
I have not; running microk8s would be similar to the guide here, right? https://github.com/canonical/microk8s
I have not used microk8s before and need to reproduce the error on my side.

Joshua-Riek, 2023-03-09:
I was able to reproduce the error on my side:
```
root@orange-pi5:~# kubectl get events
LAST SEEN   TYPE      REASON                             OBJECT            MESSAGE
4m1s        Warning   FailedNodeAllocatableEnforcement   node/orange-pi5   Failed to update Node Allocatable Limits ["kubepods"]: openat2 /sys/fs/cgroup/kubepods/cpu.weight: no such file or directory
```
However, I'm still working on a solution.

adubkov, 2023-03-09:
It is definitely running on original image:

```
orangepi@orangepi5:~$ uname -a
Linux orangepi5 5.10.110-rockchip-rk3588 #1.1.2 SMP Thu Feb 16 20:30:24 CST 2023 aarch64 aarch64 aarch64 GNU/Linux
orangepi@orangepi5:~$ microk8s status
microk8s is running
high-availability: no
  datastore master nodes: 127.0.0.1:19001
  datastore standby nodes: none
addons:
  enabled:
    ha-cluster           # (core) Configure high availability on the current node
    helm                 # (core) Helm - the package manager for Kubernetes
    helm3                # (core) Helm 3 - the package manager for Kubernetes
  disabled:
    cert-manager         # (core) Cloud native certificate management
    community            # (core) The community addons repository
    dashboard            # (core) The Kubernetes dashboard
    dns                  # (core) CoreDNS
    host-access          # (core) Allow Pods connecting to Host services smoothly
    hostpath-storage     # (core) Storage class; allocates storage from host directory
    ingress              # (core) Ingress controller for external access
    kube-ovn             # (core) An advanced network fabric for Kubernetes
    mayastor             # (core) OpenEBS MayaStor
    metallb              # (core) Loadbalancer for your Kubernetes cluster
    metrics-server       # (core) K8s Metrics Server for API access to service metrics
    minio                # (core) MinIO object storage
    observability        # (core) A lightweight observability stack for logs, traces and metrics
    prometheus           # (core) Prometheus operator for monitoring and logging
    rbac                 # (core) Role-Based Access Control for authorisation
    registry             # (core) Private image registry exposed on localhost:32000
    storage              # (core) Alias to hostpath-storage add-on, deprecated
```

Joshua-Riek, 2023-03-09:
Interesting, how did you install microk8s?

adubkov, 2023-03-09:
Yeah, I just run:

```
sudo apt update
sudo apt install snapd
sudo snap install microk8s --classic --channel=1.26
sudo usermod -a -G microk8s $USER
sudo chown -f -R $USER ~/.kube
su - $USER
```

Joshua-Riek, 2023-03-10:
I just got an older version of microk8s working (1.19), and I think that I see the issue.
Will try with a more recent version and hopefully it works.

Joshua-Riek, 2023-03-10:
Sweet, I got it running. I must pinpoint the exact changes I made to get this working. 
```
ubuntu@orange-pi5:~$ microk8s version
MicroK8s v1.26.1 revision 4567
ubuntu@orange-pi5:~$ microk8s status
microk8s is running
high-availability: no
  datastore master nodes: 127.0.0.1:19001
  datastore standby nodes: none
addons:
  enabled:
    ha-cluster           # (core) Configure high availability on the current node
    helm                 # (core) Helm - the package manager for Kubernetes
    helm3                # (core) Helm 3 - the package manager for Kubernetes
  disabled:
    cert-manager         # (core) Cloud native certificate management
    community            # (core) The community addons repository
    dashboard            # (core) The Kubernetes dashboard
    dns                  # (core) CoreDNS
    host-access          # (core) Allow Pods connecting to Host services smoothly
    hostpath-storage     # (core) Storage class; allocates storage from host directory
    ingress              # (core) Ingress controller for external access
    kube-ovn             # (core) An advanced network fabric for Kubernetes
    mayastor             # (core) OpenEBS MayaStor
    metallb              # (core) Loadbalancer for your Kubernetes cluster
    metrics-server       # (core) K8s Metrics Server for API access to service metrics
    minio                # (core) MinIO object storage
    observability        # (core) A lightweight observability stack for logs, traces and metrics
    prometheus           # (core) Prometheus operator for monitoring and logging
    rbac                 # (core) Role-Based Access Control for authorisation
    registry             # (core) Private image registry exposed on localhost:32000
    storage              # (core) Alias to hostpath-storage add-on, deprecated
```

adubkov, 2023-03-10:
Awesome! Looking forward to test it.

Joshua-Riek, 2023-03-10:
I have things to do for work tonight, so ill figure it out tomorrow morning.

adubkov, 2023-03-10:
In a mean time I was trying manually reproduce your change on existing system, but it doesn't seem to help: : https://github.com/Joshua-Riek/ubuntu-orange-pi5/commit/4a325c2431f28f86dc11f172ce8928d29b4d1500
Maybe I'm missing some steps.

Joshua-Riek, 2023-03-10:
Edit the file /boot/firmware/boot.cmd and in the bootargs string add `cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 systemd.unified_cgroup_hierarchy=0` 

Then recompile the boot script with `sudo mkimage -A arm64 -O linux -T script -C none -n "Boot Script" -d /boot/firmware/boot.cmd /boot/firmware/boot.scr`

That commit won't fix it, but this might? I have not tested

adubkov, 2023-03-10:
All right. It indeed help!
```$ microk8s status
microk8s is running
high-availability: no
  datastore master nodes: 127.0.0.1:19001
  datastore standby nodes: none
addons:
  enabled:
    ha-cluster           # (core) Configure high availability on the current node
    helm                 # (core) Helm - the package manager for Kubernetes
    helm3                # (core) Helm 3 - the package manager for Kubernetes
  disabled:
    cert-manager         # (core) Cloud native certificate management
    community            # (core) The community addons repository
    dashboard            # (core) The Kubernetes dashboard
    dns                  # (core) CoreDNS
    host-access          # (core) Allow Pods connecting to Host services smoothly
    hostpath-storage     # (core) Storage class; allocates storage from host directory
    ingress              # (core) Ingress controller for external access
    kube-ovn             # (core) An advanced network fabric for Kubernetes
    mayastor             # (core) OpenEBS MayaStor
    metallb              # (core) Loadbalancer for your Kubernetes cluster
    metrics-server       # (core) K8s Metrics Server for API access to service metrics
    minio                # (core) MinIO object storage
    observability        # (core) A lightweight observability stack for logs, traces and metrics
    prometheus           # (core) Prometheus operator for monitoring and logging
    rbac                 # (core) Role-Based Access Control for authorisation
    registry             # (core) Private image registry exposed on localhost:32000
    storage              # (core) Alias to hostpath-storage add-on, deprecated```

Joshua-Riek, 2023-03-10:
I added the changes for this in the commit fc55978c10cf07a015cdeede71631b81cb9df9a6 and will be included in the next release.

closed 2023-03-10 by Joshua-Riek