Joshua-Riek/ubuntu-rockchip #63: GPIO doesnt work

Neirno, 2023-03-10:
Hello, I got an error. Using
sudo gpio mode 2 out
sudo gpio write 2 1
Pin is not powered.
Is it possible that this problem is due to the system?

Joshua-Riek, 2023-03-10:
It could be something with the OS, I have not tested the GPIO extensively yet. But do you get an error message, or is the pin just not powered?

Neirno, 2023-03-10:
Doesn't throw an error. Just no power to the pin

Neirno, 2023-03-10:
Thank you very much!

Joshua-Riek, 2023-03-10:
I just tested blinking an LED, and it works. However, the LED is dim, and I could not tell it was on at first. 
```
root@orange-pi5:~# gpio mode 2 out
root@orange-pi5:~# gpio readall
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 |      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
 |   47 |   0 |    SDA.5 |     IN | 1 |  3 || 4  |   |        | 5V       |     |      |
 |   46 |   1 |    SCL.5 |     IN | 1 |  5 || 6  |   |        | GND      |     |      |
 |   54 |   2 |    PWM15 |    OUT | 0 |  7 || 8  | 0 | IN     | RXD.0    | 3   | 131  |
 |      |     |      GND |        |   |  9 || 10 | 0 | IN     | TXD.0    | 4   | 132  |
 |  138 |   5 |  CAN1_RX |     IN | 1 | 11 || 12 | 1 | IN     | CAN2_TX  | 6   | 29   |
 |  139 |   7 |  CAN1_TX |     IN | 1 | 13 || 14 |   |        | GND      |     |      |
 |   28 |   8 |  CAN2_RX |     IN | 1 | 15 || 16 | 1 | IN     | SDA.1    | 9   | 59   |
 |      |     |     3.3V |        |   | 17 || 18 | 1 | IN     | SCL.1    | 10  | 58   |
 |   49 |  11 | SPI4_TXD |     IN | 1 | 19 || 20 |   |        | GND      |     |      |
 |   48 |  12 | SPI4_RXD |     IN | 1 | 21 || 22 | 1 | IN     | GPIO2_D4 | 13  | 92   |
 |   50 |  14 | SPI4_CLK |     IN | 1 | 23 || 24 | 1 | IN     | SPI4_CS1 | 15  | 52   |
 |      |     |      GND |        |   | 25 || 26 | 1 | IN     | PWM1     | 16  | 35   |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
root@orange-pi5:~# gpio write 2 0
root@orange-pi5:~# gpio write 2 1
```

Joshua-Riek, 2023-03-10:
Make sure you're using the correct pin on the board, its something that I often get messed up.
[image]

Neirno, 2023-03-10:
See the situation is this: I connect the cooler to the physical pin 7 and gave it power. But it doesn't work. Is it possible that there is simply not enough power in the board? Or is it still a problem that I have a defective board?
I also tried to connect the cooler to the 3.3V power pins and everything works fine.

Joshua-Riek, 2023-03-10:
That might be the issue. What is your power supply rated for? You should be using something at least 5V/4A. 
Also, what are you trying to power from pin 7? A simple LED?

Neirno, 2023-03-11:
I connect the LED strip to this pin. I use the 5V 4A power supply that came with orange pi 5
Tried to connect different devices to different pins. As a result, everything worked from ordinary pins that only supply power, but not from gpio

Joshua-Riek, 2023-03-11:
That is strange. I only have a few singular LED's laying around that work, maybe the GPIO pin can not supply enough power for the whole strip?

Neirno, 2023-03-11:
Maybe you're right.
I will buy one LED in the coming days and write here
Thank you very much for your time!

Joshua-Riek, 2023-03-11:
No problem! I would also try the official Orange Pi 5 OS, maybe there is some bug with my version.

Joshua-Riek, 2023-03-18:
Closing this, as I have not been able to reproduce this issue, GPIO works for me. 
If there are still issues, please let me know.

closed 2023-03-18 by Joshua-Riek

Neirno, 2023-03-22:
All works, thx)