Joshua-Riek/ubuntu-rockchip #65: Can not run  XORG on V1.9

ypatriot, 2023-03-11:
I tried to Install the fresh V1.9 image and it boot to Gnome-Wayland succesfully, 
but when I choose to login with xorg ubuntu it can not load ubuntu-xorg and back to login screen. so I back to use gnome-wayland.
Then I install ubuntu-mate-desktop ( my favourite Desktop) then configure lightdm as  default DM, after rebooted it cannot load lightdm loginscreen so I reconfigure the GDM3 as default and the V1.9 image only can gnome wayland and only GDM3 as DM, so I revert to V1.7

Joshua-Riek, 2023-03-11:
Due to driver compatibility, I made some changes to Mesa and no longer use the x11 blob in favor of wayland. 
Ill take a log as I know Xorg is very popular.

Joshua-Riek, 2023-03-11:
I just did some regression testing, the issue is a result of the commit a3fa4b539075c396b30011b4242d734c9eeb846f.

Joshua-Riek, 2023-03-11:
Weird, this also happens in v1.8. Thank you for finding this, I have been so focused on Gnome that i forgot to test Xorg at all.

Joshua-Riek, 2023-03-11:
I found the commit where Xorg breaks 68554a0013155b284d31b7114d1232d59250bc40.

Joshua-Riek, 2023-03-11:
The issue is the xserver packages i'm using. Manually updating them will make xorg work again.
```
sudo apt-get install xserver-common xserver-xorg-core xserver-xorg-dev xserver-xorg-legacy
```
I do not yet know how this will affect wayland or if the version of xserver has some patches required for wayland to work properly.

Joshua-Riek, 2023-03-11:
I fixed this with commit b83a395f37d113043f310f70c42debec6a054ee5. I will be reuploading v1.9 to include this change. Thanks for reporting this issue!

closed 2023-03-11 by Joshua-Riek

ypatriot, 2023-03-12:
Thanks for your great efforts