Joshua-Riek/ubuntu-rockchip #68: reboot not working

liuyangCV, 2023-03-13:
Hi,  I found that the ubuntu cannot reboot. OS: v1.8 desktop.

liuyangCV, 2023-03-13:
Both "sudo reboot" and Button on UI are not working. I'm wondering if it's related to hardware?  When i reboot, it just worked like power off, cannot power on.

Joshua-Riek, 2023-03-13:
I see, do you know if you are using a 5V/4A power supply? Sometimes if using something below 5V/4A booting and rebooting can be weird.

ferrellsl, 2023-03-13:
If he's been playing with his CPU governor that could be the culprit.  I experienced the same problem after setting my governor to a low of 2Ghz and a high of 2.4Ghz.  Setting my governor to a maximum of 2Ghz and a low of 800Mhz with On-Demand performance made the problem go away.

Joshua-Riek, 2023-03-13:
There is a systemd service that sets the CPU and GPU governer to performance on boot. The services can be disabled, though.
```
sudo systemctl disable cpu-governor-performance
sudo systemctl disable gpu-governor-performance
```

liuyangCV, 2023-03-14:
Thank you for your response. I'm using a 5V/4A power supply, and the problem still existed when I disabled cpu and gpu performance mode.

Joshua-Riek, 2023-03-14:
Very strange. Do you have an SSD inserted? Or are you just using an SD card?

liuyangCV, 2023-03-14:
I'm just using a SD card. I will try another SD card and see what will happen.

Joshua-Riek, 2023-03-14:
Try the latest release v1.9. I double-checked on my Pi5 and it can reboot normally.

liuyangCV, 2023-03-16:
I have tried v1.9 and the official ubuntu os, both not working. Now I sent it back to orange pi office to check the hardward. Thanks for your feedback!

closed 2023-03-16 by Joshua-Riek