Joshua-Riek/ubuntu-rockchip #75: Audio/ALSA Not Working

ByteAtATime, 2023-03-17:
Hi again, I'm rather new to this whole thing but I think the audio is not working? When I go into the "Sound" in Settings everything is blank.

Joshua-Riek, 2023-03-17:
Audio should be working fine through HDMI, are you using display port?

ByteAtATime, 2023-03-17:
I'm using HDMI, this is what the settings screen looks like:
[image]
As you can see, all the fields are blank.

Joshua-Riek, 2023-03-17:
Weird, I guess the pulseaudio service stopped running. 
Try this, open a terminal and enter the command `systemctl --user restart pulseaudio`. Do not run as root, run as your current user.

ByteAtATime, 2023-03-17:
Hmm, doesn't look like anything happened? Fields are still blank, tried to reboot and rerun, still nothing.

Joshua-Riek, 2023-03-17:
Do you recall if the audio was working at any point? Also if you have an extra SD card, I would try a fresh install on it and check the sound settings. Maybe something was downloaded that conflicts with the audio.

closed 2023-03-26 by Joshua-Riek

carlesoriol, 2023-05-30:
I'm blocked in the same situation. In my case in ubuntu I can see the sound applet for few seconds then it disappears and I got the same empty devices screen

Joshua-Riek, 2023-05-30:
Do you know what version you downloaded?

If on an older version (below v1.13) try the below commands:
```
git clone -b v1.13 https://github.com/Joshua-Riek/ubuntu-orange-pi5.git
sudo cp -r ubuntu-orange-pi5/overlay/usr/share/alsa /usr/share
sudo cp -r ubuntu-orange-pi5/overlay/usr/share/pulseaudio /usr/share
```

darkcircle, 2023-09-08:
I also have the similar problem after upgrade ubuntu.
Top-right speaker icon being with network and power icon in the GNOME desktop 
is disappeared about 10 seconds later after logged in.
I saw repeating message as below when I saw speaker icon is gone away.

[  214.127071] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2

Of course, I did not met any problem before reboot after upgrade (which situation is needed to restart).
I should bake an another memory chip by using clean image, 
and I also should avoid to upgrade until this issue's key reason is revealed clearly.

Thank you in all the time!

Joshua-Riek, 2023-09-08:
> I also have the similar problem after upgrade ubuntu. Top-right speaker icon being with network and power icon in the GNOME desktop is disappeared about 10 seconds later after logged in. I saw [...]

What board are you using? And what audio device?

darkcircle, 2023-09-09:
I'm using Orange Pi 5 Plus 16GB model.

Joshua-Riek, 2023-09-09:
I did some testing on the latest release, and the audio is working fine, even after a system update.

darkcircle, 2023-09-09:
and I also found strange issue that disappearing(disabling) audio icon
when I have set disable autologin of lightdm (If I would like to have login manually by typing a password).
(I just comment out two autologin option line in /etc/lightdm/lightdm.conf.d/22-orangepi-autologin.conf
 and set greeter-show-manual-login to true in /etc/lightdm/lightdm.conf.d/11-orangepi.conf)

Changing username, password and/or hostname is not affect this problem.

Joshua-Riek, 2023-09-09:
Are you using this distro? Those files do not exist and i believe they are from the orangepi ubuntu build?

lhommev, 2023-10-17:
Hi, I have the same issue on nanoPC T6. Although everything else looks fine.

Joshua-Riek, 2023-10-17:
The audio jack does not work on the T6, but hdmi audio will work.

lhommev, 2023-10-17:
Well it does with the FriendlyElec image. So i suspect there is some misconfiguration.

Joshua-Riek, 2023-10-17:
That does not mean much. The kernel is a mess filled with hacks, and I believe they are on an older kernel version from Rockchip. 

My guess is a pulseaudo / alsa config is required.

lhommev, 2023-10-17:
Thanks, i ll try that way. I'll let you know my results

lhommev, 2023-10-17:
Ok, it took me a while but i made it. On the friendlyElec image, there are 2 configs:
/usr/share/alsa/init/00main
`CARDINFO{driver}=="realtek_rt5616-", INCLUDE="rt5616", GOTO="init_end"`

and /usr/share/alsa/init/rt5616
```

# Configuration for rt5616 driver

# playback
CTL{reset}="mixer"
CTL{name}="DAC MIXL INF1 Switch", CTL{value}="on"
CTL{name}="DAC MIXR INF1 Switch", CTL{value}="on"
CTL{name}="Stereo DAC MIXL DAC L1 Switch", CTL{value}="on"
CTL{name}="Stereo DAC MIXR DAC R1 Switch", CTL{value}="on"

CTL{name}="DAC1 Playback Volume", CTL{value}="159,159"
CTL{name}="HP Playback Volume", CTL{value}="31,31"

CTL{name}="OUT MIXL DAC L1 Switch", CTL{value}="on"
CTL{name}="OUT MIXR DAC R1 Switch", CTL{value}="on"
CTL{name}="HPO MIX DAC1 Switch", CTL{value}="off"
CTL{name}="HPO MIX HPVOL Switch", CTL{value}="on"
CTL{name}="HPVOL Playback Switch", CTL{value}="on,on"

# record
# CTL{name}="ADC Capture Switch", CTL{value}="on,on"
CTL{name}="RECMIXL BST1 Switch", CTL{value}="on"
CTL{name}="RECMIXR BST1 Switch", CTL{value}="on"
CTL{name}="Stereo1 ADC MIXL ADC1 Switch", CTL{value}="on"
CTL{name}="Stereo1 ADC MIXR ADC1 Switch", CTL{value}="on"

CTL{name}="IN1 Boost Volume", CTL{value}="1"
CTL{name}="ADC Boost Volume", CTL{value}="0,0"
CTL{name}="ADC Capture Volume", CTL{value}="47,47"

RESULT="true", EXIT="return"

```

lhommev, 2023-10-17:
sounds great ;)