Joshua-Riek/ubuntu-rockchip #77: Fix iwlwifi-ty-a0-gf-a0 and add linux-firmware deb package

Mreddi1, 2023-03-18:
Hello! Firstly, thank you so much for your hard work! Your Ubuntu image, much, much better than OrangePi provides. 

But I have a few questions:
1. Would it be possible to resolve the problem with Intel WIFI 6 AX210 module driver? It is supported by the 5.10 kernel version (https://wireless.wiki.kernel.org/en/users/Drivers/iwlwifi). But occasionally, when the linux-firmware updates, I need to run a small command and fix the wi-fi connection. I use this command: ``sudo mv /usr/lib/firmware/iwlwifi-ty-a0-gf-a0.pnvm  /usr/lib/firmware/iwlwifi-ty-a0-gf-a0.bak`` (https://askubuntu.com/questions/1360175/intel-wifi-6-ax210-wifi-not-working-after-update)
2. Would it be possible to add ``linux-firmware_****-0ubuntu1_all.deb`` package into the image? Because this package provides the support with Intel WIFI 6 AX210 module.

Joshua-Riek, 2023-03-18:
I see that the kernel module for Intel WIFI 6 AX210 exists in the kernel. I wonder if it just needs to be loaded as I include linux-firmware by default.

You could try a fresh install, enter the below command then reboot your system. That may work.
```
echo "iwlwifi" | sudo tee -a /etc/modules-load.d/modules.conf
```

Mreddi1, 2023-03-18:
@Joshua-Riek Got it. Thank you! I will try it on the next time.

Joshua-Riek, 2023-03-18:
I don't think there is a fix on this kernel version for your issue about Intel WIFI 6 AX210 wifi breaking when linux-firmware updates. 
You could hold the package with `sudo apt-mark hold linux-firmware` to prevent it from updating, so you never need to worry about your wifi randomly breaking.

Joshua-Riek, 2023-03-20:
I was able to get my hands on an Intel WIFI 6 AX210 module and I added commit 1e97a41a696ee91dd214f785fc1f3932261cf7b0 to enable wifi for this device right out of the box. It can still break if the linux-firmware package is updated by the user. I don't see any fix for this besides updating to a more recent kernel version, but the only supported kernel version is 5.10.110 for this device.