Joshua-Riek/ubuntu-rockchip #78: Wifi connectivity.

Akira2i, 2023-03-19:
I've attached an m.2 to e key adapter to use an Intel wifi module ax210ngw. The driver and device is detected but there is no wifi tab in the networks tab. Thanks for the help. Newest image being used.

Joshua-Riek, 2023-03-19:
Hey! I have this wifi module coming in sometime today, and will be able to look into this further once it arrives!

Joshua-Riek, 2023-03-19:
There is a [well-documented](https://askubuntu.com/questions/1360175/intel-wifi-6-ax210-wifi-not-working-after-update) issue where updating the `linux-firmware` package will break wifi on this device. This is not something I can fix, but I can make sure this wifi module works out of the box before a user updates the `linux-firmware` package, if it even has an update available.

If the wifi module stops working after an update, enter the below command:
```
sudo mv /usr/lib/firmware/iwlwifi-ty-a0-gf-a0.pnvm  /usr/lib/firmware/iwlwifi-ty-a0-gf-a0.bak
```

Joshua-Riek, 2023-03-20:
I added commit 1e97a41a696ee91dd214f785fc1f3932261cf7b0 to enable wifi for this device right out of the box. It can still break if the linux-firmware package is updated. I don't see any fix for this besides updating to a more recent kernel version, but the only supported kernel version is 5.10.110 for this device.

closed 2023-03-20 by Joshua-Riek