Joshua-Riek/ubuntu-rockchip #79: Bluetooth firmware not loading for Realtek adapter

fvillanustre, 2023-03-19:
When using a Realtek usb dongle with wifi/bluetooth, the kernel fails to load rtl_bt/rtl8821c_fw.bin with error -2. Wifi works fine but bluetooth doesn't.

The specific error is: 
bluetooth hci0: Direct firmware load for rtl_bt/rtl8821c_fw.bin failed with error -2
Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8821c_fw.bin not found

This is with the latest Jammy image.

Joshua-Riek, 2023-03-19:
It looks like the firmware file is missing, but rtl8821c_fw.bin should be included by default. 
Could you give me the output of the command `ls -la /usr/lib/firmware/rtl_bt`.

fvillanustre, 2023-03-19:
Of course! The file is there, it just doesn't load (kernel symbol mismatches, perhaps?):

total 1168
drwxr-xr-x   2 root root   4096 Mar 11 22:48 .
drwxr-xr-x 100 root root  20480 Mar 11 23:04 ..
-rw-r--r--   1 root root  38764 Mar 11 22:48 rtl8192ee_fw.bin
-rw-r--r--   1 root root  37904 Mar 11 22:48 rtl8192eu_fw.bin
-rw-r--r--   1 root root  24548 Mar 11 22:48 rtl8723a_fw.bin
-rw-r--r--   1 root root     68 Mar 11 22:48 rtl8723b_config.bin
-rw-r--r--   1 root root  45048 Mar 11 22:48 rtl8723b_fw.bin
-rw-r--r--   1 root root     68 Mar 11 22:48 rtl8723bs_config.bin
lrwxrwxrwx   1 root root     29 Jan 26 08:10 rtl8723bs_config-OBDA0623.bin -> rtl8723bs_config-OBDA8723.bin
-rw-r--r--   1 root root     64 Mar 11 22:48 rtl8723bs_config-OBDA8723.bin
-rw-r--r--   1 root root  46236 Mar 11 22:48 rtl8723bs_fw.bin
-rw-r--r--   1 root root     10 Mar 11 22:48 rtl8723d_config.bin
-rw-r--r--   1 root root  52080 Mar 11 22:48 rtl8723d_fw.bin
-rw-r--r--   1 root root     47 Mar 11 22:48 rtl8723ds_config.bin
-rw-r--r--   1 root root  48988 Mar 11 22:48 rtl8723ds_fw.bin
-rwxr-xr-x   1 root root      6 Mar 11 22:48 rtl8761a_config.bin
-rw-r--r--   1 root root  74488 Mar 11 22:48 rtl8761a_fw.bin
-rw-r--r--   1 root root     25 Mar 11 22:48 rtl8761b_config.bin
-rw-r--r--   1 root root  44204 Mar 11 22:48 rtl8761b_fw.bin
-rw-r--r--   1 root root      6 Mar 11 22:48 rtl8761bu_config.bin
-rw-r--r--   1 root root  42088 Mar 11 22:48 rtl8761bu_fw.bin
-rw-r--r--   1 root root  40520 Mar 11 22:48 rtl8812ae_fw.bin
lrwxrwxrwx   1 root root     19 Jan 26 08:10 rtl8821a_config.bin -> rtl8821c_config.bin
-rw-r--r--   1 root root  37420 Mar 11 22:48 rtl8821a_fw.bin
-rw-r--r--   1 root root     10 Mar 11 22:48 rtl8821c_config.bin
-rw-r--r--   1 root root  51700 Mar 11 22:48 rtl8821c_fw.bin
-rw-r--r--   1 root root     14 Mar 11 22:48 rtl8822b_config.bin
-rw-r--r--   1 root root  51176 Mar 11 22:48 rtl8822b_fw.bin
-rw-r--r--   1 root root     33 Mar 11 22:48 rtl8822cs_config.bin
-rw-r--r--   1 root root  62040 Mar 11 22:48 rtl8822cs_fw.bin
-rw-r--r--   1 root root      6 Mar 11 22:48 rtl8822cu_config.bin
-rw-r--r--   1 root root  59332 Mar 11 22:48 rtl8822cu_fw.bin
-rw-r--r--   1 root root      6 Mar 11 22:48 rtl8852au_config.bin
-rw-r--r--   1 root root  49844 Mar 11 22:48 rtl8852au_fw.bin
-rw-r--r--   1 root root      6 Mar 11 22:48 rtl8852bu_config.bin
-rw-r--r--   1 root root  55276 Mar 11 22:48 rtl8852bu_fw.bin
-rw-r--r--   1 root root      6 Mar 11 22:48 rtl8852cu_config.bin
-rw-r--r--   1 root root 106960 Mar 11 22:48 rtl8852cu_fw.bin
-rw-r--r--   1 root root     55 Mar 11 22:48 rtlbt_config
-rw-r--r--   1 root root  44504 Mar 11 22:48 rtlbt_fw
-rw-r--r--   1 root root  51364 Mar 11 22:48 rtlbt_fw_new

Joshua-Riek, 2023-03-19:
Strange that it can't load. I might need to order an adapter to debug further. What is the exact dongle you are using? I just checked, and RTL8821CU is enabled in the kernel. However, I don't know kernel programming well if this is a kernel issue.
I can recommend [this](https://www.amazon.com/dp/B099K2FD1J?psc=1&ref=ppx_yo2ov_dt_b_product_details) dongle for wifi and Bluetooth, it works great on this system.

fvillanustre, 2023-03-19:
Great! I'll order that one. In the meantime, this is the [one](https://www.amazon.com/Bluetooth-Adapter-Wireless-External-Receiver/dp/B07YDFZWT8) that I have.

fvillanustre, 2023-03-19:
And in case this helps, this is how lsusb sees my device: 

```
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Wireless, Driver=btusb, 480M
        ID 0bda:c820 Realtek Semiconductor Corp. 
    |__ Port 1: Dev 2, If 1, Class=Wireless, Driver=btusb, 480M
        ID 0bda:c820 Realtek Semiconductor Corp. 
    |__ Port 1: Dev 2, If 2, Class=Vendor Specific Class, Driver=rtl8821cu, 480M
        ID 0bda:c820 Realtek Semiconductor Corp. 

```

Joshua-Riek, 2023-03-19:
Oh, I forgot there was a similar issue #38. Unplugging and replugging the dongle should work for the moment.
But taking another look I think the firmware is trying to load rtl_bt/rtl8821c_fw.bin too quickly. I have an idea of how to fix this.

fvillanustre, 2023-03-19:
I can confirm that unplugging and re-plugging the adapter addresses the issue. It may be due to attempting to load that firmware too quickly at boot time, as you said.

Joshua-Riek, 2023-03-19:
This should now be fixed with the commit e948474. I added the missing firmware to the initrd and now it works on my end.

If you would like to test on your running system, you can add the below code to the file `/usr/share/initramfs-tools/hooks/rtl-bt`.
```bash
#!/bin/sh

PREREQ=""
prereqs()
{
    echo "$PREREQ"
}

case $1 in
prereqs)
    prereqs
    exit 0
    ;;
esac

. /usr/share/initramfs-tools/scripts/functions
. /usr/share/initramfs-tools/hook-functions

files=$(find /usr/lib/firmware/rtl_bt)
for file in $files; do
    copy_exec "${file}" /usr/lib/firmware/rtl_bt
done

exit 0
```
Then mark the file as executable and update the initrd.
```
sudo chmod +x /usr/share/initramfs-tools/hooks/rtl-bt
sudo update-initramfs -u
```
After a system reboot it should work without issue!

fvillanustre, 2023-03-19:
I can confirm that the suggested change fixed the issue. Now the system boots properly with the USB adapter plugged in and both, wifi and bluetooth work without issues. Thanks!!!

Joshua-Riek, 2023-03-19:
Glad to hear that. Thanks for reporting the issue!

closed 2023-03-19 by Joshua-Riek