Joshua-Riek/ubuntu-rockchip #82: SSH login fails

rahul-kurup, 2023-03-21:
Using latest build v1.10

I've tried both `root` and `ubuntu` users with password `ubuntu`. But its failing:

[image]

Joshua-Riek, 2023-03-21:
Using the server or desktop image?

rahul-kurup, 2023-03-21:
server

Joshua-Riek, 2023-03-21:
Got it, taking a look now.

Joshua-Riek, 2023-03-21:
Weird, ssh login fails on my side too. I think its an issue with cloud-init

rahul-kurup, 2023-03-21:
I was about to flash desktop version, was waiting for your reply. You may close this ticket and test it at a later period of time

Joshua-Riek, 2023-03-21:
Thanks for reporting this issue! I forgot to update the fs label for cloud-init, and it has been fixed with commit dde33ec485de22d042fcd615770f6fb2b848a750. I'll start a build on my server and update the release in a few hours.

closed 2023-03-21 by Joshua-Riek

artem-zinnatullin, 2023-03-24:
Huh, actually ssh with `ubuntu` user and `ubuntu` password worked! But ssh with `root` user doesn't :)

Joshua-Riek, 2023-03-24:
root login through ssh is disabled by default for security reasons.