Joshua-Riek/ubuntu-rockchip #83: How to update an OS

Singular112, 2023-03-21:
Hi. How to update an OS that was already burned to the SD card instead of a new install ?

New versions releasing frequently and it is impossible to reinstall the system every time and configure everything from scratch.

Joshua-Riek, 2023-03-21:
I have had a lot of people asking the same question, and creating an update process will be the next thing I do. The process will include updating the kernel, bootloader, firmware, and packages. This will contain most changes between versions, I will need to do a lot of testing to get this right.

stuartcamerondeakin, 2023-03-22:
It's still early days, I'd honestly rather you spend your time on anything that's still likely to yield big improvements to the system, incremental updates are going to more useful once there's less of a tinkerer-focused and more stable branch focused audience.

Joshua-Riek, 2023-03-23:
You make a great point, I have multiple improvements in the works, and it's difficult to keep up with the rapidly developing changes around this board when long-term stability and updates are in mind. Thank you for the feedback.

Joshua-Riek, 2023-03-25:
I have a small update. I made some mirrors of liujianfeng1994's PPAs, which contain multiple packages related to hardware acceleration. This will allow me to test changes made to liujianfeng1994's PPA before updating my mirror and distributing those changes to users. However, I will need some time to implement it properly, I need to re-work how I implement hardware acceleration and make sure it does not produce any new issues.

stuartcamerondeakin, 2023-03-25:
Sounds awesome, appreciate the time you're putting into this!

Joshua-Riek, 2023-03-27:
All mesa and multimedia-related packages are now available through a PPA with the commit 54325f4031900933dfacffe9be4807698488a57d. I will clean up the commit later when I have time. I will be working on packaging the kernel, bootloader, and firmware at a later date now that I have a process in place for distributing packages.

Joshua-Riek, 2023-04-15:
I'm happy to say the kernel and firmware are fully packaged now on my PPA. I need to add the bootloader and a few more packages, but distributing package updates should be as simple as an apt upgrade in the next release. 

https://launchpad.net/~jjriek/+archive/ubuntu/orangepi5/+packages

ferrellsl, 2023-04-15:
> I'm happy to say the kernel and firmware are fully packaged now on my PPA. I need to add the bootloader and a few more packages, but distributing package updates should be as simple as an apt upgrade in the next release.
> 
> https://launchpad.net/~jjriek/+archive/ubuntu/orangepi5/+packages

Does this mean that if I'm running release v1.11 that I can now upgrade to v1.12 by doing a sudo apt update followed by a sudo apt upgrade?

Joshua-Riek, 2023-04-15:
I'm not sure about that yet. I've had to change a few essential boot files for the update process to work.

Joshua-Riek, 2023-04-21:
I'm happy to announce that [v1.13](https://github.com/Joshua-Riek/ubuntu-orange-pi5/releases/tag/v1.13) is now released, and users will be able to receive kernel, firmware, and bootloader updates through apt. However, this is not compatible with any previous release.

closed 2023-04-21 by Joshua-Riek

tanmaster, 2023-05-08:
>I'm happy to announce that [v1.13](https://github.com/Joshua-Riek/ubuntu-orange-pi5/releases/tag/v1.13) is now released, and users will be able to receive kernel, firmware, and bootloader updates through apt. However, this is not compatible with any previous release.

thank you for your work, this is great!