Joshua-Riek/ubuntu-rockchip #84: Random mac address on reboot

rahul-kurup, 2023-03-22:
Sys info -
I'm on v1.10. 
OS is running from NVMe drive.
Internet is running via Ethernet. 

Problem -
It seems like on every restart the MAC address keeps changing. Is there a way to keep it consistent (use the hardware one)?

ironiris, 2023-03-22:
ubuntu@ubuntu:~$ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.60  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::20a6:7a54:392:5fcc  prefixlen 64  scopeid 0x20<link>
        **ether be:39:29:4a:d1:04  txqueuelen 1000  (Ethernet)**
        RX packets 9337360  bytes 10933187375 (10.9 GB)
        RX errors 0  dropped 77168  overruns 0  frame 0
        TX packets 1460200  bytes 3196324641 (3.1 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 86

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 5139  bytes 553229 (553.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 5139  bytes 553229 (553.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlx90de8093713f: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 90:de:80:93:71:3f  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ubuntu@ubuntu:~$ sudo reboot
[sudo] ubuntu 암호:
login as: ubuntu
ubuntu@192.168.1.60's password:
Welcome to Ubuntu 22.04.2 LTS (GNU/Linux 5.10.110-rockchip-rk3588 aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

 * Strictly confined Kubernetes makes edge and IoT secure. Learn how MicroK8s
   just raised the bar for easy, resilient and secure K8s cluster deployment.

   https://ubuntu.com/engage/secure-kubernetes-at-the-edge

 * Introducing Expanded Security Maintenance for Applications.
   Receive updates to over 25,000 software packages with your
   Ubuntu Pro subscription. Free for personal use.

     https://ubuntu.com/pro

Expanded Security Maintenance for Applications is not enabled.

0 updates can be applied immediately.

24 additional security updates can be applied with ESM Apps.
Learn more about enabling ESM Apps service at https://ubuntu.com/esm

2 updates could not be installed automatically. For more details,
see /var/log/unattended-upgrades/unattended-upgrades.log

Last login: Thu Mar 23 08:18:54 2023 from 192.168.1.131
ubuntu@ubuntu:~$ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.60  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::20a6:7a54:392:5fcc  prefixlen 64  scopeid 0x20<link>
        **ether be:39:29:4a:d1:04  txqueuelen 1000  (Ethernet)**
        RX packets 270  bytes 50636 (50.6 KB)
        RX errors 0  dropped 38  overruns 0  frame 0
        TX packets 149  bytes 28083 (28.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 86

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 135  bytes 11869 (11.8 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 135  bytes 11869 (11.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlx90de8093713f: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 90:de:80:93:71:3f  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
------------------
In my case, it is the same mac address even after rebooting.

rahul-kurup, 2023-03-23:
I restarted thrice and this is how it looks. Everytime the mac changed, thus changing the IP address

[image]

Joshua-Riek, 2023-03-23:
I just tested, and my system always uses the same mac and IP address. Are you able to set a static IP address through the settings menu?

rahul-kurup, 2023-03-23:
I don't think I can assign static mac via settings in the OS itself. Static IP needs to be set from router interface. I have assigned static IPs to my laptop and phone. And they work fine even on reboot. But for this guy, it keeps on changing the mac address, thus pinning an IP becomes difficult

rahul-kurup, 2023-03-23:
that didn't work but this profile changed on restart

[image]

Joshua-Riek, 2023-03-23:
Strange, do you know if you have had this issue before on any other versions of the OS?

rahul-kurup, 2023-03-23:
I've started using your OS since 1.10 only. But as far as I remember I haven't faced this issue on Armbian or the Android version

Joshua-Riek, 2023-03-23:
Do you have a wifi dongle that works? I would like to know if it has the same issue.

Maybe this would work, edit `/etc/NetworkManager/conf.d/30-randomize-mac-address.conf` with the below content and reboot:
```
[connection-mac-randomization]
ethernet.cloned-mac-address=permanent
wifi.cloned-mac-address=permanent
```

rahul-kurup, 2023-03-23:
I have tried this before opening the issue. But it didn't work. Also I don't have a separate wifi dongle

Joshua-Riek, 2023-03-23:
I'll be able to test more tomorrow, and I hope to reproduce this issue on my system so I can debug the problem further.

rahul-kurup, 2023-03-23:
Just for context. I've flashed Armbian Jammy (Desktop) now

[image]

A new mac gets assigned at first (not sure why hardware one is not picked by default); but the new one stays persistent across reboots. Thus ssh and everything keeps on working without any issue

Joshua-Riek, 2023-03-25:
I have not been able to reproduce this with the server or desktop image. Like Armbian, I get a new mac address on the first boot, which will persist across reboots.

rahul-kurup, 2023-03-26:
Dunno what else can be done for this, as I'm stuck with the different macs right now. Anyways thanks for checking this issue out

rahul-kurup, 2023-03-26:
@Joshua-Riek Are you using Ethernet or Wifi?

Joshua-Riek, 2023-03-26:
I tested both ethernet and wifi.

rahul-kurup, 2023-03-26:
I reflashed the system and everything seems to be working fine except for root login (this is desktop edition):

[image]

Joshua-Riek, 2023-03-26:
Glad to hear! Root login is disabled by default for security reasons.

closed 2023-03-26 by rahul-kurup

smjltd, 2023-07-23:
had the same issue, seems to be an issue with orange pi 5 HW. 

Someone on reddit suggested the following solution.
[Orange Pi 5 - Mac address keeps changing](https://www.reddit.com/r/OrangePI/comments/14sleyi/comment/jqza65e/)

i just simply used usb type c dongle with ethernet.

rahul-kurup, 2023-07-23:
@smjltd that someone is me 😅 As you said seems like an hw thing, but the solution which I've provided there works fine so far.

smjltd, 2023-07-29:
i guess this was the issue 
[Issue 274](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/274)

Joshua-Riek, 2023-07-29:
That issue was introduced in v1.20 because of a bootloader update from orangepi.