Joshua-Riek/ubuntu-rockchip #85: High refresh rate output issue

jcp-os, 2023-03-23:
High refresh rate settings are yielding green garbled image when beings set to resolutions higher than 60hz

jcp-os, 2023-03-23:
Issue is at: 120hz 4K
Monitor: Eve Spectrum 4k 144hz
Connection: HDMI 2.1

Joshua-Riek, 2023-03-23:
I see, can you provide more information about what happened? Do you see green garbled green images in your web browser, or is the entire desktop messed up?

jcp-os, 2023-03-23:
It was in settings, when switching display refresh rates. The whole display is green and garbled overlay.

Joshua-Riek, 2023-03-23:
Sometimes there is a weird issue where the display can go pink/green sometimes and look garbled, to fix it, you need to unplug the HDMI and then plug it back in. However, I have a 4k 120Hz TV I can try later today.

Joshua-Riek, 2023-03-24:
I see what you mean. I set my TV to 100Hz and had a small number of artifacts, then set it to 120Hz, and the screen went crazy with artifacts. I will need to try the official OS from Orange Pi and see if it has the same issue, if it does the problem might be out of my hands.

Joshua-Riek, 2023-03-27:
This issue also happens with the official Orange Pi OS and I don't see any documentation saying 4k 120Hz is officially supported on the Orange Pi 5. I don't know enough to dive deeper and fix this issue.

closed 2023-03-27 by Joshua-Riek

jcp-os, 2023-03-27:
Resolved the issue by using a different power supply.

Joshua-Riek, 2023-03-27:
Oh, it worked with a different power supply? Strange.

jcp-os, 2023-03-27:
Definitely strange as the orange pi is only pulling 7-10W according to my cable. 

Changed the brick from 5V 3A to 5V 4A. and now the screen is stable at 4K 120hz. ran youtube 4k stream for 3 hours and never experienced the glitch.

Thanks for troubleshooting though!