Joshua-Riek/ubuntu-rockchip #86: Kernel messages flooding logs

fvillanustre, 2023-03-23:
A particular kernel error message is flooding the system logs in the latest build:

`Mar 02 23:31:38 kitchen-opi kernel: [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351`

It seems that amiberry is triggering this behavior, but I don't think this is exclusive to amiberry.

I haven't observed any other impact on the system, other than this odd behavior.

Joshua-Riek, 2023-03-23:
Hmm, I have seen this come up before. I don't know enough about the kernel to fix the error properly. But I'm sure that I can suppress it so logs are not flooded for the time being.

fvillanustre, 2023-03-23:
I think that suppressing it is the best way forward. The specific error message was introduced into the kernel when they added VOP2 support, back in 2021, so I'm not sure it will be addressed any time soon.

Joshua-Riek, 2023-03-28:
I found that playing a full-screen video with MPV can generate this message. However, the message has been suppressed in commit 97357f19eb8abc475deb50c4f4f67368783cbaea. Thanks for reporting this issue!

closed 2023-03-28 by Joshua-Riek

rpardini, 2023-08-25:
Hey Joshua, affected by the same, on rkr5.1 on Armbian. I guess this patch was lost? It's not in your fork either.

Joshua-Riek, 2023-08-25:
Good catch. It looks like the patch was lost. I think it would be good to introduce it into the rkr5.1 branch, but I would comment out the entire DRM_ERROR line.
```diff
diff --git a/drivers/gpu/drm/rockchip/rockchip_drm_vop2.c b/drivers/gpu/drm/rockchip/rockchip_drm_vop2.c
index 3a4b87cd5..12f81ba6c 100644
--- a/drivers/gpu/drm/rockchip/rockchip_drm_vop2.c
+++ b/drivers/gpu/drm/rockchip/rockchip_drm_vop2.c
@@ -2259,7 +2259,7 @@ static bool rockchip_vop2_mod_supported(struct drm_plane *plane, u32 format, u64
                return true;
 
        if (!rockchip_afbc(plane, modifier) && !rockchip_tiled(plane, modifier)) {
-               DRM_ERROR("Unsupported format modifier 0x%llx\n", modifier);
+               //DRM_ERROR("Unsupported format modifier 0x%llx\n", modifier);
 
                return false;
        }
```