Joshua-Riek/ubuntu-rockchip #88: Freeze system issue

Lahutsin, 2023-03-25:
When you are copying any large file tracker-miner-fs-3 scan it and up CPU  is going to freeze the system on NVMe. For resolving this issue I change the setting for:

```
echo -e "\nHidden=true\n" | sudo tee --append /etc/xdg/autostart/tracker-extract.desktop /etc/xdg/autostart/tracker-miner-apps.desktop /etc/xdg/autostart/tracker-miner-fs.desktop /etc/xdg/autostart/tracker-miner-user-guides.desktop /etc/xdg/autostart/tracker-store.desktop > /dev/null

# Interval in days to check whether the filesystem is up to date in the database. 0 forces crawling anytime, -1 forces it only after unclean shutdowns, and -2 disables it entirely
**gsettings set org.freedesktop.Tracker.Miner.Files crawling-interval -2  # Default: -1**
# Set to false to completely disable any file monitoring
**gsettings set org.freedesktop.Tracker.Miner.Files enable-monitors false # Default: true**
```

more: https://askubuntu.com/questions/346211/tracker-store-and-tracker-miner-fs-eating-up-my-cpu-on-every-startup

Joshua-Riek, 2023-03-25:
I prefer not to disable default features in Ubuntu, but if it causes the system to freeze, it seems reasonable to disable it. 
How large of a file where you copying? I want to reproduce this myself before making any changes.

Lahutsin, 2023-03-25:
~2Gb - 10 files

Joshua-Riek, 2023-03-25:
Thanks, ill take a look!

ironiris, 2023-03-25:
I often copy more than 10 gigabytes of files from the web to nvme or sdcard, but I've never had a problem.

ferrellsl, 2023-03-25:
> I often copy more than 10 gigabytes of files from the web to nvme or sdcard, but I've never had a problem.

Same here.  I've ported quite a few games to the Pi 5 with game content/DLC in excess of 10GB.  I copied the files via USB and SD card to my nvme drive without any issues.

Joshua-Riek, 2023-03-25:
@Lahutsin are you using a proper 5v/4a power supply? Maybe your system is freezing / crashing due to power issues as tracker-miner-fs uses the entire CPU to index.

Lahutsin, 2023-03-26:
I inspected my power supply: 5V-3A / 9V-3A / 12V-3A
Also, I have 5V-4A 
I'll make some tests for it and make some report if I'll get something special

Joshua-Riek, 2023-03-26:
Try 5V-4A, I would suspect it to fix the issue.

closed 2023-03-31 by Joshua-Riek