Joshua-Riek/ubuntu-rockchip #91: ID 0bda:b82c Realtek Semiconductor Corp. 802.11ac NIC with a problem on the first boot with the device connected

ironiris, 2023-03-30:
The above device is a dongle with simultaneous WIFI and BT support,
When it boots up, it freezes, even though there are no log lines.

There was no problem before version 1.11.
The power supply is also 5V 4A.

If I complete the setup without it, and then connect and use it, there is no problem.

Joshua-Riek, 2023-03-30:
I will take a look. Thanks for letting me know about this.

rahul-kurup, 2023-03-30:
@ironiris On a side-note, can you share the product link from where I can buy this. I've been looking for a dongle which supports both Wifi and Bt simultaneously.

ironiris, 2023-03-30:
> @ironiris On a side-note, can you share the product link from where I can buy this. I've been looking for a dongle which supports both Wifi and Bt simultaneously.

https://a.aliexpress.com/_m0eqhYi
 It was the best choice when I bought it, but now there are faster models at a similar price.

ironiris, 2023-03-30:
> I will take a look. Thanks for letting me know about this.

 It's good to handle it slowly. Only need to unplug and boot the device once.

Joshua-Riek, 2023-03-31:
I forgot, I had this issue a while ago here https://github.com/Joshua-Riek/ubuntu-orange-pi5/issues/38#issuecomment-1450681050. I think this is a kernel issue, and not really something I can fix. Also I tested v1.10 and v1.9 of my releases and have the same issue. We can open an issue on the official Orange Pi kernel or build system GitHub page.

Joshua-Riek, 2023-04-07:
I finally fixed this with the commit 36a0007b99d67f68af971c6d872b46ed36009e7a. I added usb_modeswitch and the associated udev rules into the initrd.

closed 2023-04-07 by Joshua-Riek