Joshua-Riek/ubuntu-rockchip #98: Rename .img.sha to .img.xz.sha256 #enhancement

morencia, 2023-04-02:
Correct file naming convention: Please rename ".img.sha" files to ".img.xz.sha256", as they contain SHA-256 hash for ".img.xz" (not for ".img"). Thanks

Joshua-Riek, 2023-04-02:
I think that I will remove the `.img.xz.sha` files in the next release. The release assets would get cluttered with the upcoming Orange Pi 5B and Orange Pi 5Plus boards. I still plan to provide the sha256 hash for each image, but in a text box, not a file.

I have an example screenshot of what I'm planning:
[image]

morencia, 2023-04-03:
That approach will force me to update my automation scripts and parse the full new release github page to gather checksums, but whatever is easier for you (while clear and consistent) will be fine :) Thanks

Joshua-Riek, 2023-04-03:
I drafted up a new release format that I like as shown below. I will be keeping the `img.sha` files and renaming them to `img.sha256` as you suggested. 

[image]

morencia, 2023-04-03:
Very nice-looking, but "img.xz.sha256" would be technically correct (unless you hash the images before compression) ;-) Thanks

Joshua-Riek, 2023-04-03:
Thanks for pointing that out!

Joshua-Riek, 2023-04-04:
The `.img.sha` files have been changed to `.img.xz.sha256` with the commit ce93b358c4f10cd16d2c1c2b821de695fd3338d0.
Thanks for bringing this to my attention!