Joshua-Riek/ubuntu-rockchip #99: Netflix logging out

cantalupo555, 2023-04-02:
Whenever I start some content it logs out, directs the page to:
https://www.netflix.com/clearcookies

Sometimes it presents these errors and tries to log out:
ui3012
m7121-1331

All other streaming works perfectly in Chromium.

ironiris, 2023-04-02:
Have you changed the string of the user-agent?

cantalupo555, 2023-04-02:
> Have you changed the string of the user-agent?

No, clean installation of the system.
No changes were made.

rahul-kurup, 2023-04-03:
@cantalupo555 Netflix will keep on logging out, **unless you change your user agent to Chrome OS while browsing**. 
Something like this should work for you:

- Install [User-Agent Switcher](https://chrome.google.com/webstore/detail/user-agent-switcher-for-c/djflhoibgkdhkhhcedjiklpkjnoahfmg) extension
- Get latest [ChromeOS user agent](https://www.whatismybrowser.com/guides/the-latest-user-agent/chrome-os); use **Aarch64 on Chrome OS** and set it up in this extension.
- Switch to the above user agent.
- Now navigate to https://netflix.com. Login and watch whatever you want.
- Bonus: Install [Netflix1080p](https://chrome.google.com/webstore/detail/netflix-1080p/cankofcoohmbhfpcemhmaaeennfbnmgp) extension for better viewing experience. Check all the options shown in the extension settings

Joshua-Riek, 2023-04-03:
I will probably add this in the next few days. I have been very busy with my job and planning for future changes.

Big-BenG, 2023-04-03:
How can I get the latest ChromeOS user agent ? Many thanks

cantalupo555, 2023-04-03:
@rahul-kurup 
[image]

Where do I configure the extension?

[image]

cantalupo555, 2023-04-03:
Got it!

[image]

[image]

closed 2023-04-04 by cantalupo555