Joshua-Riek/ubuntu-rockchip #101: Screen over usbc doesn't work

HSTe, 2023-04-04:
I tried connecting screen over usbc on an adapter. It works over some other images I have tried for orange pi 5.
In dmesg I get this:
[ 1386.363133] dw-dp fde50000.dp: failed to probe DP link: -110
[ 1399.905624] dw-dp fde50000.dp: failed to probe DP link: -110
[ 1402.910871] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp0
[ 1402.928670] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 1920x1080p60, type: 11(if:800) for vp0 dclk: 148500000
[ 1402.928792] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx_ropll_cmn_config bus_width:16a8c8 rate:490000
[ 1402.929057] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
[ 1402.929157] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] dclk_out0 div: 0 dclk_core0 div: 2
[ 1402.929223] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx_ropll_cmn_config bus_width:16a8c8 rate:1485000
[ 1402.929488] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
[ 1402.929495] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] set dclk_vop0 to 148500000, get 148500000

Joshua-Riek, 2023-04-04:
I have been using an HDMI to usbc adapter, and it works fine for display on my end. Try to reboot and ensure nothing is plugged into the white USB port.

HSTe, 2023-04-04:
I got it working if I unplug the monitor I have on the hdmi output. But it will not work wih both monitors connected

Joshua-Riek, 2023-04-04:
I know this has come up before in issue #32. I developed an initial fix to prevent hardware acceleration from breaking when simultaneously plugging in HDMI and USB-C displays.

However, my initial fix is more of a hack, which is probably why this occurs. In addition, I'm unfamiliar with how the HDMI and USB-C interact with the system at a kernel level. So there is not much more I can do unless I find some other project with this issue and they have a solution.

closed 2023-04-18 by Joshua-Riek

laamalif, 2023-06-22:
Version: 1.18

Running headless mode, no HDMI or USB-C display cable attached.

[429865.384370] dw-dp fde50000.dp: timeout waiting for AUX reply
[429865.651035] dw-dp fde50000.dp: timeout waiting for AUX reply
[429865.917707] dw-dp fde50000.dp: timeout waiting for AUX reply
[429866.184378] dw-dp fde50000.dp: timeout waiting for AUX reply
[429866.451050] dw-dp fde50000.dp: timeout waiting for AUX reply
[429866.717804] dw-dp fde50000.dp: timeout waiting for AUX reply
[429866.984484] dw-dp fde50000.dp: timeout waiting for AUX reply
503356.555297] dw-dp fde50000.dp: timeout waiting for AUX reply
[503356.821935] dw-dp fde50000.dp: timeout waiting for AUX reply
[503357.088603] dw-dp fde50000.dp: timeout waiting for AUX reply

Joshua-Riek, 2023-06-22:
I recall seeing that before. I had it disabled at some point in the kernel. I will need to look back and see if it was removed.