Joshua-Riek/ubuntu-rockchip #104: HDMI Audio Stops Working After Upgrade

hubby4, 2023-04-08:
I'm having a similar issue as #75.

I'm using version 1.11.

The problem occurs for me whenever I do "sudo apt update && sudo apt upgrade". Even if a I try "sudo apt update && sudo apt dist-upgrade", the HDMI audio still stops working. The only workaround I've found is to upgrade a program individually to avoid triggering whatever upgrade is responsible for disabling the audio.

Natisfaction, 2023-04-08:
Same problem here, even with aptitude I had no luck

HSTe, 2023-04-08:
I have the same issue. This is what was upgraded:
Start-Date: 2023-04-08  13:04:13
Commandline: apt upgrade
Requested-By: test (1000)
Upgrade: libldb2:arm64 (2:2.4.4-0ubuntu0.22.04.1, 2:2.4.4-0ubuntu0.22.04.2), xse
rver-xorg-core:arm64 (2:21.1.3-2ubuntu2.8, 2:21.1.3-2ubuntu2.9), vim:arm64 (2:8.
2.3995-1ubuntu2.4, 2:8.2.3995-1ubuntu2.5), xxd:arm64 (2:8.2.3995-1ubuntu2.4, 2:8
.2.3995-1ubuntu2.5), libgdm1:arm64 (42.0-1ubuntu7, 42.0-1ubuntu7.22.04.1), liblouis20:arm64 (3.20.0-2ubuntu0.1, 3.20.0-2ubuntu0.2), alsa-ucm-conf:arm64 (1.2.6.3-1ubuntu1.4, 1.2.6.3-1ubuntu1.5), vim-common:arm64 (2:8.2.3995-1ubuntu2.4, 2:8.2.3995-1ubuntu2.5), libwbclient0:arm64 (2:4.15.13+dfsg-0ubuntu1, 2:4.15.13+dfsg-0ubuntu1.1), liblouis-data:arm64 (3.20.0-2ubuntu0.1, 3.20.0-2ubuntu0.2), libsmbclient:arm64 (2:4.15.13+dfsg-0ubuntu1, 2:4.15.13+dfsg-0ubuntu1.1), gnome-settings-daemon-common:arm64 (42.1-1ubuntu2.1, 42.1-1ubuntu2.2), libmagickcore-6.q16-6-extra:arm64 (8:6.9.11.60+dfsg-1.3ubuntu0.22.04.2, 8:6.9.11.60+dfsg-1.3ubuntu0.22.04.3), python3-louis:arm64 (3.20.0-2ubuntu0.1, 3.20.0-2ubuntu0.2), xserver-xorg-legacy:arm64 (2:21.1.3-2ubuntu2.8, 2:21.1.3-2ubuntu2.9), gdm3:arm64 (42.0-1ubuntu7, 42.0-1ubuntu7.22.04.1), libc6:arm64 (2.35-0ubuntu3.1+1, 2.35-0ubuntu3.1+3), locales:arm64 (2.35-0ubuntu3.1+1, 2.35-0ubuntu3.1+3), gnome-settings-daemon:arm64 (42.1-1ubuntu2.1, 42.1-1ubuntu2.2), xserver-common:arm64 (2:21.1.3-2ubuntu2.8, 2:21.1.3-2ubuntu2.9), xwayland:arm64 (2:22.1.1-1ubuntu0.5, 2:22.1.1-1ubuntu0.6), libmagickwand-6.q16-6:arm64 (8:6.9.11.60+dfsg-1.3ubuntu0.22.04.2, 8:6.9.11.60+dfsg-1.3ubuntu0.22.04.3), vim-tiny:arm64 (2:8.2.3995-1ubuntu2.4, 2:8.2.3995-1ubuntu2.5), distro-info-data:arm64 (0.52ubuntu0.2, 0.52ubuntu0.3), linux-firmware:arm64 (20220329.git681281e4-0ubuntu3.11, 20220329.git681281e4-0ubuntu3.12), python3-ldb:arm64 (2:2.4.4-0ubuntu0.22.04.1, 2:2.4.4-0ubuntu0.22.04.2), gir1.2-gdm-1.0:arm64 (42.0-1ubuntu7, 42.0-1ubuntu7.22.04.1), vim-runtime:arm64 (2:8.2.3995-1ubuntu2.4, 2:8.2.3995-1ubuntu2.5), libmagickcore-6.q16-6:arm64 (8:6.9.11.60+dfsg-1.3ubuntu0.22.04.2, 8:6.9.11.60+dfsg-1.3ubuntu0.22.04.3), libc-dev-bin:arm64 (2.35-0ubuntu3.1+1, 2.35-0ubuntu3.1+3), xserver-xephyr:arm64 (2:21.1.3-2ubuntu2.8, 2:21.1.3-2ubuntu2.9), samba-libs:arm64 (2:4.15.13+dfsg-0ubuntu1, 2:4.15.13+dfsg-0ubuntu1.1), libc-bin:arm64 (2.35-0ubuntu3.1+1, 2.35-0ubuntu3.1+3), libc-devtools:arm64 (2.35-0ubuntu3.1+1, 2.35-0ubuntu3.1+3), imagemagick-6-common:arm64 (8:6.9.11.60+dfsg-1.3ubuntu0.22.04.2, 8:6.9.11.60+dfsg-1.3ubuntu0.22.04.3), libc6-dbg:arm64 (2.35-0ubuntu3.1+1, 2.35-0ubuntu3.1+3), libc6-dev:arm64 (2.35-0ubuntu3.1+1, 2.35-0ubuntu3.1+3)
End-Date: 2023-04-08  13:06:30

Joshua-Riek, 2023-04-08:
The package `alsa-ucm-conf` seems to have broken some of the config settings I have set for this device.

Here is a quick fix:
```bash
git clone https://github.com/Joshua-Riek/ubuntu-orange-pi5.git
sudo cp -r ubuntu-orange-pi5/overlay/usr/share/alsa /usr/share
sudo cp -r ubuntu-orange-pi5/overlay/usr/share/pulseaudio /usr/share
```
Then reboot your system, and the sound should be working again.

HSTe, 2023-04-08:
Thanks. That was fixed fast

hubby4, 2023-04-08:
Thanks Josh. That workaround works.

Joshua-Riek, 2023-04-08:
There is a config file that gets overwritten with the package update. I might need to prevent the alsa-ucm-conf package from updating so this does not happen. 

I'll devise a permanent solution in the next release. Thank you for bringing this to my attention.

Natisfaction, 2023-04-08:
Thank you so much!

develperbayman, 2023-04-08:
actually man god bless your hard working heart for sure! .......just know we all appreciate the crap out of all your hard work!!!

Joshua-Riek, 2023-04-10:
I came up with a temporary fix for this issue in commit 3e96808f2672bb09e3173c0bbb6e97ebc7acde6b. I have a permanent solution in the works that will come at a later date.

cantalupo555, 2023-04-10:
> The package `alsa-ucm-conf` seems to have broken some of the config settings I have set for this device.
> 
> Here is a quick fix:
> 
> ```shell
> git clone https://github.com/Joshua-Riek/ubuntu-orange-pi5.git
> sudo cp -r ubuntu-orange-pi5/overlay/usr/share/alsa /usr/share
> sudo cp -r ubuntu-orange-pi5/overlay/usr/share/pulseaudio /usr/share
> ```
> 
> Then reboot your system, and the sound should be working again.

Thank you.

Joshua-Riek, 2023-04-21:
I patched the alsa-ucm-conf package to fix this issue. It will be available on my PPA in the next release.
https://launchpad.net/~jjriek/+archive/ubuntu/orangepi5

closed 2023-04-21 by Joshua-Riek